gnuton/asuswrt-merlin.ng #220: tuf ax3000  Chinese firmware font overlay problem

qqw1231, 2022-07-30:
[image]

Chinese characters are superimposed where I use red marks, and there are some cards when each page of the router is opened

gnuton, 2022-07-30:
[image]
is this still wrong?
Is the localization looking any better on stock firmware?

qqw1231, 2022-07-31:
> [image] is this still wrong? Is the localization looking any better on stock firmware?

There is something wrong with the typesetting of ax3000 on this page, and each click on each page is a little slow

gnuton, 2022-08-02:
is it slower than the stock firmware? 
I still do not get what/s wrong, can you be more explicit?

wspruce, 2022-08-24:
> I still do not get what/s wrong, can you be more explicit?
I guess they means the newline before every last character in each menu item. It breaks chinese words apart and looks a little bit wierd. The page you show has less tags so there is enough space.