gnuton/asuswrt-merlin.ng #221: Stuck on web history tab

fuloskop, 2022-07-30:
**Router Model Affected**
Models: DSL-AX82U

**Firmware Version Affected**
[386.5_2-gnuton0]

**Is this bug present in upstream Merlin releases too?**
I dont know

**Describe the bug**
If you open AdaptiveQoS_WebHistory.asp sometimes router interface stuck. You cant do nothing. web history must be on.

**To Reproduce**
Steps to reproduce the behavior:
1. I open web history
2. After 3 4 days
3. I click web history. Page load but web_frame is empty
4. in this case you can't do anything

**Expected behavior**
I expect nothing to happen, show web history and be able to continue.

**Screenshots**
[image]
[image]

WillyTP, 2022-11-06:
Hi,
same issue on 386.08_0-gnuton1

Moreover, in my case web interface crashes.
I cannot access to it for about 10 minutes, when it restarts automatically.
This after accessing Web History page.

`Nov  6 23:11:33 NiTroZ-DSL-AX82U HTTPD: waitting 10 minitues and restart
Nov  6 23:11:33 NiTroZ-DSL-AX82U rc_service: httpds 2164:notify_rc restart_httpd
Nov  6 23:11:33 NiTroZ-DSL-AX82U custom_script: Running /jffs/scripts/service-event (args: restart httpd)
Nov  6 23:11:33 NiTroZ-DSL-AX82U DSL-AX82U: start https:8443
Nov  6 23:11:33 NiTroZ-DSL-AX82U DSL-AX82U: start https:8443
Nov  6 23:11:33 NiTroZ-DSL-AX82U DSL-AX82U: start httpd:80
Nov  6 23:11:33 NiTroZ-DSL-AX82U custom_script: Running /jffs/scripts/service-event-end (args: restart httpd)
`

gnuton, 2022-11-07:
@WillyTP 
The logs says you the web ui crashes because you have a custom script called service-event-end.
(see the last line) that's a script which doesn't belong to Merlin itself.
You can disable JFFS apps or remove the buggy script.