gnuton/asuswrt-merlin.ng #224: Wrong default section on the left side of the home screen

DervilRus, 2022-07-31:
**Router Model Affected**
Models: TUF-AX5400

**Firmware Version Affected**
386.07_2-gnuton0_beta4

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
Wrong default section on the left side of the home screen. 
On Merlin firmware, when you click the NETWORK MAP tab, the contents of the Status tab are displayed on the left. On this firmware, by default, the contents of the WIRELESS tab are shown

**To Reproduce**
Steps to reproduce the behavior:
1. Login to router
2. or Click on 'Network Map'

DervilRus, 2022-08-25:
Default Merlin
[image]

Default Gnuton
[image]

gnuton, 2022-08-25:
This is an issue of the tuff theme. I can fix that but i do have plan to
get rid of it.

DervilRus, 2022-08-25:
> This is an issue of the tuff theme. I can fix that but i do have plan to get rid of it.
> […](#)

here this problem has the same sources?
https://github.com/gnuton/asuswrt-merlin.ng/issues/222