gnuton/asuswrt-merlin.ng #225: gray-blue ASUS web interface on TUF routers

DervilRus, 2022-07-31:
**Is your feature request related to a problem? Please describe.**
NO

**Describe the solution you'd like**
I would like to be able to switch the color of the web interface to the standard gray-blue ASUS from a gaming yellow-black TUF.
If this is of course possible to implement.

timrag, 2022-08-01:
Has to be a simple file in /www doesn't it?

gnuton, 2022-08-02:
@timrag it's a few of them like CSS and some JS.
I will look better into this, but i do not know if this is something that everybody wants

yk00365, 2022-08-03:
> @timrag it's a few of them like CSS and some JS. I will look better into this, but i do not know if this is something that everybody wants

Yes，many of the tuf users want to use the asuswrt blue skin,because the tuf style is course possible to implement. @gnuton

MrRojano97, 2022-08-04:
Please if you can, add a skin selector, personally i don't like too much TUF interface instead of blue Asus router interfaces. Thank you very much for your work.

denis-g, 2022-08-05:
Maybe make sense add all specific skins for all models?

yk00365, 2022-08-07:
> Please if you can, add a skin selector, personally i don't like too much TUF interface instead of blue Asus router interfaces. Thank you very much for your work.

Sounds like a good idea.