gnuton/asuswrt-merlin.ng #226: kernel: protocol 0000 is buggy, dev archer

Karmakstylez, 2022-08-04:
**Router Model Affected**
AX82U + TUF5400

**Firmware Version Affected**
Unstable: 386.07_2-gnuton0_beta4

**Is this bug present in upstream Merlin releases too?**
I do not know.

**Describe the bug**
kernel: protocol 0000 is buggy, dev archer. Seems to be a kernel issue? I do not know what the consequences are. If it could be fixed, please do so. :)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to /Main_LogStatus_Content.asp
2. See Logs for Error

**Expected behavior**
No protocol 0000 is buggy, dev archer in logs. 

**Screenshots**
[image]

gnuton, 2022-08-04:
Hi, could you please tell me if you still have this issue after resetting the router to factory?

gnuton, 2022-10-06:
This is a known "issue" https://www.snbforums.com/threads/asx88u-general-log-spam-protocol-is-buggy.70968/