gnuton/asuswrt-merlin.ng #228: Wlan power / more rx and tx

RealSongoku, 2022-08-09:
Hi to you!
Is it posible to get this feature?

I had the r7000 and now the tuf 5400. 
R7000 with merlin i had the choice from 0-100% power level.

I think 50% where in real 100%.

Is it posssible to ad more wlan power?

Thank you
Daniel

Roki100, 2022-08-09:
> Hi to you! Is it posible to get this feature?
> 
> I had the r7000 and now the tuf 5400. R7000 with merlin i had the choice from 0-100% power level.
> 
> I think 50% where in real 100%.
> 
> Is it posssible to ad more wlan power?
> 
> Thank you Daniel

im pretty sure you cant go above 100% nor change the mW power (as its hardcoded in encrypted CFE(bootloader)) but you could try changing the region in nvram if thats what you mean

RealSongoku, 2022-08-11:
Ok i understand but with the original software the signal was stronger. But with the latestet stable the signal is not so strong.

Any chance to get this work ?

Roki100, 2022-08-11:
> Ok i understand but with the original software the signal was stronger. But with the latestet stable the signal is not so strong.
> 
> Any chance to get this work ?

i am not sure of what do you mean

RealSongoku, 2022-08-11:
Wlan with original firmware was stronger then with gnutons firmware.

RealSongoku, 2022-08-14:
Hi gnu!
On the same place sometimes my wlan signal gets lost with handy. (With new firmware)
With original firmware the signal is stable.

So no chance to get a stronger wlan signal?

gnuton, 2022-08-14:
@RealSongoku not sure what are the differences you see. once again if you have some numbers that would help, but in any case    there should not be difference between my builds and stock when it comes to wireless.
Modern wireless interfaces have usually power setting enabled which make your handy loose the connection to wireless to save battery.
That's an option you can disable in th firmware from the wireless page

RealSongoku, 2022-08-15:
Whats the name of the option in the firmware?

Ok maybe my phone have an isue. 

But its not bosdible do add some more mw to rx and tx ?

gnuton, 2022-08-18:
Enable WMM APSD: Enable or Disable WMM APSD (Automatic Power Save Delivery).

RealSongoku, 2022-08-19:
Hi.
I know that i can change it to performance (i have it)
But maybe its more possible? :)
Thank you

vichaos, 2022-10-16:
You can easily check the wireless transmission power using this command
`nvram dump|grep -E "wl*._txpower"` 

For each key/value that doesn't matched the stock firmware you could overwrite it with this command

`nvram set wl0_txpower=XXX
nvram set wl1_txpower=XXX
nvram set wl_txpower=XXX

nvram commit
service restart_wireless
`

If the above instruction had solved the issue. You might want to put them in https://github.com/RMerl/asuswrt-merlin.ng/wiki/User-scripts to survive after the router restarted

Roki100, 2022-10-16:
@vichaos from what i know this shouldnt work/take any effect on newer routers such as the tuf ax5400 due to regulatory domains

vichaos, 2022-10-24:
> @vichaos from what i know this shouldnt work/take any effect on newer routers such as the tuf ax5400 due to regulatory domains
> but if it somehow makes any difference then it sounds interesting

Yes, you were right and I also think that using the highest transmission power on any non-backhaul band(s) doesn't benefit much but it leads to higher CCI, higher noise, sticky clients, bad roaming and many more since most of client devices usually have lower transmission power. 

Personally, I'm using  XT8 and this tx power setting seems to provide the best speed and response time across all devices. 

[image]