gnuton/asuswrt-merlin.ng #230: DHCP Error on DSL-AX82U with latest Firmware

FreoPhil, 2022-08-10:
**Router Model Affected**
Models: (eg: DSL-AX82U/RT-AX82U/...)
DSL-AX82U

**Firmware Version Affected**
386.07_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
After updating to the latest Merlin firmware release I could not get reconnected to my Aussie Broadband VDSL connection. The issue was showing as “Your ISP’s DHCP Does Not Function”. The link was showing as up but no internet. Worked with Aussie Broadband tech support on this error for 90 minutes with no luck, reset everything and even factory reset with no luck. Tried suggested fixes in configurations as shown in multiple discussions on line and no luck. In the end had to roll back to previous version to get a connection again. Would appreciate any advice or suggestions. 

**To Reproduce**
Steps to reproduce the behavior:
Install the latest firmware update

**Expected behavior**n.

**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section

gnuton, 2022-08-11:
Hi, do you have super loop? This is a know issue I am working on a fix

gnuton, 2022-08-11:
duplicate of https://github.com/gnuton/asuswrt-merlin.ng/issues/204
Please follow the issue above for updates

closed 2022-08-11 by gnuton

FreoPhil, 2022-08-11:
Hi, no I use Aussie Broadband and worked on the issue with them for 90 minutes without success. Has this been fixed?

gnuton, 2022-08-12:
@FreoPhil please follow the other issue since it's a duplicate of this

FreoPhil, 2022-08-16:
Tried the beta you sent and still had same issue with internet unable to connect to Aussie Broadband. Problem is as I have reported earlier.

gnuton, 2022-08-16:
@FreoPhil is the stock firmware working for you?
What about the previous version?

FreoPhil, 2022-08-16:
Hi, I have rolled back to 386.5_2 and it works fine. At this stage I am going to stick with this version, as it is stable and works well with my ZenWifi AX units. I tried the beta today you had sent in the link to me the other day.

Sorry but I need a stable connection.

gnuton, 2022-08-16:
Can you please tell me again your ISP provider? 
thanks

FreoPhil, 2022-08-17:
www.aussiebroadband.com.au<http://www.aussiebroadband.com.au>

They are probably one of the best in terms of technical support in Australia.

Regards

Philip White
m - +61 (0) 409 044 509