gnuton/asuswrt-merlin.ng #235: "lighttpd/15525: potentially unexpected fatal signal 11" errors

patsio, 2022-08-14:
**Router Model Affected**
Models: AC68U
**Firmware Version Affected**
386.07_2-gnuton1 Stable

**Is this bug present in upstream Merlin releases too?**
I don't know

**Describe the bug**
"lighttpd/15525: potentially unexpected fatal signal 11" errors every 2-3 hours to my log file

**To Reproduce**
Steps to reproduce the behavior:
Enable the AiCloud web server 
**Expected behavior**
According to RMerlin, these errors actually indicate that lighttpd (the AiCloud web server) has crashed.
Disabling the Cloud Web Server, these entries are gone the last 12 hours.
**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section

gnuton, 2022-08-15:
Confirmed

> Aug 15 21:49:21 kernel: lighttpd/4989: potentially unexpected fatal signal 11.
> Aug 15 21:49:21 kernel: Pid: 4989, comm:             lighttpd
> Aug 15 21:49:21 kernel: CPU: 1    Tainted: P             (2.6.36.4brcmarm #1)
> Aug 15 21:49:21 kernel: PC is at 0x2a100
> Aug 15 21:49:21 kernel: LR is at 0x1af78
> Aug 15 21:49:21 kernel: pc : [<0002a100>]    lr : [<0001af78>]    psr: 20000010
> Aug 15 21:49:21 kernel: sp : bed803d8  ip : 00000000  fp : bed803e4
> Aug 15 21:49:21 kernel: r10: bed80de8  r9 : 00011de8  r8 : 00010130
> Aug 15 21:49:21 kernel: r7 : bed80f4d  r6 : 00000004  r5 : bed80e74  r4 : 0005e620
> Aug 15 21:49:21 kernel: r3 : 6f685365  r2 : 00000000  r1 : 002280d9  r0 : 6f685365
> Aug 15 21:49:21 kernel: Flags: nzCv  IRQs on  FIQs on  Mode USER_32  ISA ARM  Segment user
> Aug 15 21:49:21 kernel: Control: 10c53c7d  Table: 9b09004a  DAC: 00000015

gstrauss, 2022-09-13:
What version of lighttpd is running?
What is the lighttpd config and lighttpd modules?  (`lighttpd -f /etc/lighttpd/lighttpd.conf -p`)

If reproducible, please set in lighttpd.conf `debug.log-request-header = "enable"` to capture request headers to error log (before crash).

gnuton, 2022-09-13:
I was able to reproduce this once but after that this didn't happen anymore, but I do need to check it again.
@patsio are you still getting this error?

wkngau219, 2024-11-28:
Hi, I have the same problem.

Models: AX82U V2
Firmware Version Affected
3004.388.8_2-gnuton1
Is this bug present in upstream Merlin releases too?
I am not sure.  But I can use AICloud in AX56U (using upstream Merlin firmware)

Describe the bug:
AICloud cannot be used:
[AICloud-NoLogin](https://github.com/user-attachments/assets/d19e7b25-476d-4051-b857-7731b1221e2b)

Similar log can be found in syslog:
[syslog.txt](https://github.com/user-attachments/files/17942417/syslog.txt)

The above log should be able to reproduce everytime when you access the AICloud url.

This is the lighttpd.conf when running the command `lighttpd -f /tmp/lighttpd.conf -p`:
[lighttpd.conf.txt](https://github.com/user-attachments/files/17942416/lighttpd.conf.txt)

It is noted that lighttpd error log can be found (in /tmp/lighttpd/err.log):
[err.log](https://github.com/user-attachments/files/17942415/err.log), it looks like some config-key in `lighttpd.conf` cannot be recognized (e.g. `smbdav.auth_ntlm`).

I have taken a brief look at the source tree, it seems that the file `release/src/router/lighttpd-1.4.39/src/mod_smbdav.c` exists in upstream merlin but not in this repo.  I am not sure if it is relevant.

wkngau219, 2024-11-28:
@gstrauss Thanks for your advise.  I have turned off the aicloud service now.

@gnuton I have some more findings:
I have checked the repo `release/src/router/lighttpd-1.4.39/prebuild`:
`RT-AX82U` should use `arm_6750hnd`
`RT-AX82U_V2` should use `arm_6756hnd`

This is the list of prebuild files in `release/src/router/lighttpd-1.4.39/prebuild/arm_6750hnd` (i.e. for RT-AX82U):
[image]

This is the list of prebuild files in `release/src/router/lighttpd-1.4.39/prebuild/arm_6756hnd` (i.e. for RT-AX82U_V2):
[image]

This is the list of lighttpd module files in my router (RT-AX82U_V2):
[image]

It looks like the prebuild module files in my router (RT-AX82U_V2) are using those for RT-AX82U, which is incorrect.  So I suspect there may be misconfiguration when building lighttpd.  Could you kindly have a check on it?

> > Is this bug present in upstream Merlin releases too?
> 
> The stack trace looks to be in ASUS proprietary code. The bugs are very likely still present.
> 
> I will be intentionally somewhat vague here: I am a lighttpd developer and I sent a list of security concerns to ASUS many months ago (earlier this year 2024). The response was underwhelming. I **strongly** discourage you from exposing any ASUS services to the internet.

gstrauss, 2024-11-28:
@wkngau219: good information.

> Could you kindly have a check on it?

I am sorry; I can not.  A while ago, I had reached out to ASUS to attempt to work with them, but they declined.

Everything in /prebuild/ is prebuilt.  I do not have access to the source or to build those binaries.

Perhaps @gnuton might be able to point you in the right direction for where to report this?  Good luck!