gnuton/asuswrt-merlin.ng #236: Support for RT-AX92U

gnuton, 2022-08-15:
This thread is about supporting RT-AX92U
Feel free to ask questions here. 

Status:
- [] received GPL
- [] received development unit
- [] Merged GPL to feature-branch
- [] Initial testing & Bugfixing phase
- [] Public pre-alphas
- [] Alpha released
- [] collecting testing reports  & Bugfixing phase
- [] Beta released
- [] collecting testing reports  & Bugfixing phase
- [] Stable released

ETAs:
alpha: TBD
beta: TBD
stable: TBD

nichtsnutz0815, 2022-08-17:
Thanks a lot for those news!

kilroyone, 2022-08-19:
Cool!! Thanks!

gnuton, 2022-08-23:
Looking for some testers.. I i will release soon the first public alpha

black-da-bull, 2022-08-23:
i'll volunteer... test the ai 
mesh

gnuton, 2022-08-24:
pre-alpha image available for testing.: https://output.circle-artifacts.com/output/job/397e1b5d-94df-4aa9-98c6-8ed8fa51f8e6/artifacts/0/workspace/release/RT-AX92U_386.8_0_ubi.w

riboyama, 2022-08-24:
Nice, great work! Will test/flash it this weekend, very excited

pldoom, 2022-08-24:
Awesome! Have been waiting for this quite long! :) Thank you, will test

FlourHome, 2022-08-25:
Naisu! Tried the firmware for some hours, and it seems to work fine for me :D The only problem I found was that the "WiFi Radar" does not work "http://router.asus.com/WiFi_Insight.asp" :( I only get 404'ed :(

nichtsnutz0815, 2022-08-25:
Great...thanks a lot. Just installed and tested a little bit.

techtype2c, 2022-08-27:
Installed on a mesh system of 2 routers, working great for two days! Seems to be an improvement in mesh functioning, connection stability and gui response speed. Thank you so much - an amazing alpha!!!

techtype2c, 2022-08-27:
Seeing this in the log one or two times per hour:
Aug 27 11:19:16 kernel: CPU: 0 PID: 6499 Comm: roamast Tainted: P           O    4.1.51 #2
Aug 27 11:19:16 kernel: Hardware name: Broadcom-v8A (DT)
Aug 27 11:19:16 kernel: task: ffffffc012bd20c0 ti: ffffffc010168000 task.ti: ffffffc010168000
Aug 27 11:19:16 kernel: PC is at 0xf6e8b6a4
Aug 27 11:19:16 kernel: LR is at 0xf6e57858
Aug 27 11:19:17 kernel: pc : [<00000000f6e8b6a4>] lr : [<00000000f6e57858>] pstate: 60000030
Aug 27 11:19:17 kernel: sp : 00000000f6e13cb0
Aug 27 11:19:17 kernel: x12: 00000000ffffffff 
Aug 27 11:19:17 kernel: x11: 00000000f6e141ac x10: 0000000000000000 
Aug 27 11:19:17 kernel: x9 : 00000000f6f517a8 x8 : 00000000f6f53000 
Aug 27 11:19:17 kernel: x7 : 00000000f6e15920 x6 : 0000000000000021 
Aug 27 11:19:17 kernel: x5 : 00000000001c9d16 x4 : 0000000000000000 
Aug 27 11:19:17 kernel: x3 : 0000000000000000 x2 : 0000000000000001 
Aug 27 11:19:17 kernel: x1 : 0000000000000020 x0 : 00000000fffffff8 
Aug 27 11:19:41 roamast: ROAMING Start...

techtype2c, 2022-08-27:
The error above occurs when I walk around the house or yard and my iPhone 13 tries to roam

techtype2c, 2022-08-27:
Rebooted the mesh system with the gui and observed that when anything tries to roam, that error occurs resulting in a crash and restart of the roaming function

techtype2c, 2022-08-28:
I am using wired backhaul so all 3 bands are available and I use try-band smart connect. Disabling Roaming Assistant for all 3 bands eliminates the error above (as one might expect)

techtype2c, 2022-08-28:
tri-band

valuxin, 2022-08-29:
Here is some bug with PSM. The only device connected right now is a PC with Intel AX200. Later this week I'll test it with my other devices. My router config is simple: DualWAN, WLAN without Smart Connect with manual channels, no VPNs and QoS is used.
> Aug 29 10:15:38 acsd: acs_set_chspec: 0x1909 (11u) for reason APCS_CSTIMER
Aug 29 10:15:38 kernel: wl0: chanspec: 0x1909 (11u)
Aug 29 10:15:39 acsd: eth5: selected_chspec is 0x1909 (11u)
Aug 29 10:15:39 acsd: eth5: Adjusted channel spec: 0x1909 (11u)
Aug 29 10:15:39 acsd: eth5: selected channel spec: 0x1909 (11u)
Aug 29 10:15:39 acsd: eth5: txop channel select: Performing CSA on chspec 0x1909
Aug 29 10:15:41 kernel: wl0: wlc_bmac_suspend_mac_and_wait: waited 83000 uS and MI_MACSSPNDD is still not on.
Aug 29 10:15:41 kernel: wl0: reason = mac suspend failure at 203056 seconds. corerev 42 ucode revision 1570.181 features 0x0426
Aug 29 10:15:41 kernel: psmdebug 0x00ff8edc phydebug 0x1 macctl 0x84120402 maccmd 0x4
Aug 29 10:15:41 kernel: psm_brc 0x0000 psm_brc_1 0x0020 M_UCODE_DBGST 0x3
Aug 29 10:15:41 kernel: brwk_0 0x1184 brwk_1 0x0000 brwk_2 0x0020 brwk_3 0x3103 brwk_4 0x0000
Aug 29 10:15:41 kernel: ifsstat      0x1ea0 ifsstat1     0xf0   txectl       0x4000 txestat      0x401  
Aug 29 10:15:41 kernel: rxestat1     0x6000 rxestat2     0xb40  rcv_frmcnt   0x0    rxe_rxcnt    0x6    
Aug 29 10:15:41 kernel: wepctl       0x0    
Aug 29 10:15:41 kernel: aqmfifordy   0x0    wepstat      0x0    wep_ivloc    0x24   wep_psdulen  0x52   
Aug 29 10:15:41 kernel: PC :
Aug 29 10:15:41 kernel: R0: 0xff8edd   0xff8ee0   0xfe8ede   0xc98ede  
Aug 29 10:15:41 kernel: R0: 0xc98ede   0xff8edc   0xff8ee0   0xfe8ede  
Aug 29 10:15:41 kernel: R0: 0xec8ede   0xc98ede   0xff8edc   0xff8ee0  
Aug 29 10:15:41 kernel: R0: 0xc98ede   0xff8ee0   0xec8ede   0xc98ede  
Aug 29 10:15:41 kernel: R0: 0xff8ee0   0xfe8ede   0xec8ede   0xff8edd  
Aug 29 10:15:41 kernel: R0: 0xfe8ede   0xec8ede   0xff8edd   0xff8edc  
Aug 29 10:15:41 kernel: R0: 0xff8edd   0xff8edc   0xff8ee0   0xec8ede  
Aug 29 10:15:41 kernel: R0: 0xff8edc   0xff8ee0   0xfe8ede   0xc98ede  
Aug 29 10:15:41 kernel: R0: 0xff8edc   0xff8ee0   0xec8ede   0xc98ede  
Aug 29 10:15:41 kernel: R0: 0xc98ede   0xff8edc   0xff8ee0   0xec8ede  
Aug 29 10:15:41 kernel: R0: 0xff8ee0   0xec8ede   0xc98ede   0xff8edc  
Aug 29 10:15:41 kernel: R0: 0xc98ede   0xff8edc   0xff8ee0   0xec8ede  
Aug 29 10:15:41 kernel: R0: 0xec8ede   0xff8edd   0xff8edc   0xff8ee0  
Aug 29 10:15:41 kernel: R0: 0xff8ee0   0xfe8ede   0xec8ede   0xff8edd  
Aug 29 10:15:41 kernel: R0: 0xec8ede   0xff8edd   0xff8edc   0xfe8ede  
Aug 29 10:15:41 kernel: R0: 0xff8ee0   0xec8ede   0xc98ede   0xff8edc  
Aug 29 10:15:41 kernel: phydebug :
Aug 29 10:15:41 kernel: 0x1        0x1        0x1        0x1       
Aug 29 10:15:41 kernel: 0x1        0x1        0x1        0x1       
Aug 29 10:15:41 kernel: 0x1        0x1        0x1        0x1       
Aug 29 10:15:41 kernel: 0x1        0x1        0x1        0x1       
Aug 29 10:15:41 kernel: 0x1        0x1        0x1        0x1       
Aug 29 10:15:41 kernel: 0x1        0x1        0x1        0x1       
Aug 29 10:15:41 kernel: 0x1        0x1        0x1        0x1       
Aug 29 10:15:41 kernel: 0x1        0x1        0x1        0x1       
Aug 29 10:15:41 kernel: pktproc      0x101  pktproc      0x101  pktproc      0x101  pktproc      0x101  
Aug 29 10:15:41 kernel: psm stack_status : 0x10
Aug 29 10:15:41 kernel: psm stack_entries:
Aug 29 10:15:41 kernel: 0x0eda
Aug 29 10:15:41 kernel: 0x1647
Aug 29 10:15:41 kernel: 0x11d3
Aug 29 10:15:41 kernel: 0x1eb2
Aug 29 10:15:41 kernel: 0x1bfe
Aug 29 10:15:41 kernel: 0x152a
Aug 29 10:15:41 kernel: 0x0e5c
Aug 29 10:15:41 kernel: 0x05cb
Aug 29 10:15:41 kernel: PHY reg 0x1f0 = 0x101
Aug 29 10:15:41 kernel: PHY reg 0x1f0 = 0x101
Aug 29 10:15:41 kernel: PHY reg 0x1f0 = 0x101
Aug 29 10:15:41 kernel: PHY reg 0x1f0 = 0x101
Aug 29 10:15:41 kernel: PHY reg 0x1f0 = 0x101
Aug 29 10:15:41 kernel: PHY reg 0x1f0 = 0x101
Aug 29 10:15:41 kernel: PHY reg 0x1f0 = 0x101
Aug 29 10:15:41 kernel: PHY reg 0x1f0 = 0x101
Aug 29 10:15:41 kernel: PHY reg 0x1f0 = 0x101
Aug 29 10:15:41 kernel: PHY reg 0x1f0 = 0x101
Aug 29 10:15:41 kernel: PHY reg 0x16b = 0x0 (fineclk_gatectl)
Aug 29 10:15:41 kernel: PHY reg 0x403 = 0x0 (RfseqStatus0)
Aug 29 10:15:41 kernel: PHY reg 0x408 = 0xc02 (RfctrlCmd)
Aug 29 10:15:41 kernel: RADIO reg 0x16a = 0x0 (OVR14 - AFE_PU)
Aug 29 10:15:41 kernel: RADIO reg 0x8e = 0x93 (pll_adc6 - ADC_PU)
Aug 29 10:15:41 kernel: PSM_STACK_STATUS : 0x10
Aug 29 10:15:41 kernel: PSM_STACK_ENTRIES:
Aug 29 10:15:41 kernel: 0x0eda
Aug 29 10:15:41 kernel: 0x1647
Aug 29 10:15:41 kernel: 0x11d3
Aug 29 10:15:41 kernel: 0x1eb2
Aug 29 10:15:41 kernel: 0x1bfe
Aug 29 10:15:41 kernel: 0x152a
Aug 29 10:15:41 kernel: 0x0e5c
Aug 29 10:15:41 kernel: 0x05cb
Aug 29 10:15:41 kernel: wl0: PSM microcode watchdog fired at 203056 (seconds). Resetting.
Aug 29 10:15:41 kernel: wl0: reason = psm watchdog at 203056 seconds. corerev 42 ucode revision 1570.181 features 0x0426
Aug 29 10:15:41 kernel: psmdebug 0x00fe8224 phydebug 0x1 macctl 0x84160403 maccmd 0x14
Aug 29 10:15:41 kernel: psm_brc 0x0001 psm_brc_1 0x0020 M_UCODE_DBGST 0x2
Aug 29 10:15:41 kernel: brwk_0 0x1004 brwk_1 0x0002 brwk_2 0x0030 brwk_3 0x7101 brwk_4 0x0000
Aug 29 10:15:41 kernel: wlc_dump_ucode_fatal: log_done 0x4
Aug 29 10:15:41 kernel: wl0: fatal error, reinitializing, total count of reinit's[8]

gnuton, 2022-08-29:
@techtype2c thansk for reporting this. I think it make sense to open a ticket for this. Fixing this may not be easy since AiMesh is a closed source.

riboyama, 2022-08-29:
> The error above occurs when I walk around the house or yard and my iPhone 13 tries to roam

I don't think it's an error, since setting the min log level to error doesn't make it show on my AX92U.
The roamassist functionality seems to work completely fine though

riboyama, 2022-08-29:
Overall stability seems perfect (dirty flash from stock). I get 99% speeds from ISP.
I have 1 AiMesh node running AX92U stock firmware, no issues.

Can also confirm:
> Naisu! Tried the firmware for some hours, and it seems to work fine for me :D The only problem I found was that the "WiFi Radar" does not work "http://router.asus.com/WiFi_Insight.asp" :( I only get 404'ed :(

techtype2c, 2022-08-29:
> > The error above occurs when I walk around the house or yard and my iPhone 13 tries to roam
> 
> I don't think it's an error, since setting the min log level to error doesn't make it show on my AX92U. Edit: It does restart the roamassist, client disconnects on its own:
> 
> ```
> Aug 29 19:01:03 roamast: eth6: disconnect weak signal strength station [5c:87:30:b0:db:f2]
> Aug 29 19:01:03 kernel: CPU: 1 PID: 9151 Comm: roamast Tainted: P           O    4.1.51 #2
> Aug 29 19:01:03 kernel: Hardware name: Broadcom-v8A (DT)
> Aug 29 19:01:03 kernel: task: ffffffc015ba0100 ti: ffffffc014bc8000 task.ti: ffffffc014bc8000
> Aug 29 19:01:03 kernel: PC is at 0xf6c8c72e
> Aug 29 19:01:03 kernel: LR is at 0xf6c58858
> Aug 29 19:01:03 kernel: pc : [<00000000f6c8c72e>] lr : [<00000000f6c58858>] pstate: 20000030
> Aug 29 19:01:03 kernel: sp : 00000000f6c14cb0
> Aug 29 19:01:03 kernel: x12: 00000000ffffffff 
> Aug 29 19:01:03 kernel: x11: 00000000f6c151ac x10: 0000000000000000 
> Aug 29 19:01:03 kernel: x9 : 00000000f6d527a8 x8 : 00000000f6d54000 
> Aug 29 19:01:03 kernel: x7 : 00000000f6c16920 x6 : 0000000000000021 
> Aug 29 19:01:03 kernel: x5 : 00000000001c9d16 x4 : 0000000000000004 
> Aug 29 19:01:03 kernel: x3 : 0000000000000000 x2 : 0000000000000001 
> Aug 29 19:01:03 kernel: x1 : 0000000000000058 x0 : 000000000000005c 
> Aug 29 19:01:04 kernel: wl1: random key value: 30CE33708964A044052B00FB873567E2F467BCD414EA0D9E455C7E1CDEB7F8C5
> Aug 29 19:01:04 wlceventd: wlceventd_proc_event(469): eth6: Deauth_ind 5C:87:30:B0:DB:F2, status: 0, reason: Station requesting (re)association is not authenticated with responding station (9)
> Aug 29 19:01:04 wlceventd: wlceventd_proc_event(486): eth6: Disassoc 5C:87:30:B0:DB:F2, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
> Aug 29 19:01:18 roamast: ROAMING Start...
> ```
> 
> Relevant? https://www.snbforums.com/threads/smart-connect-or-aimesh-issue-ax88u-has-many-kernel-signal-11-errors-right-after-roamast-deauth-old-sta-or-disconnect-weak-signal.80388/

That thread is interesting! I'm trying just leaving Roaming Assist turned off but it will take a few days to be sure it's a good idea.

valuxin, 2022-08-30:
4 days uptime, no issues so far (except WiFi Radar page 404 error). Thanks, very nice alpha version:)

cjnickisherxcii, 2022-08-31:
Been waiting a long time for AX-92U support! Thanks for bringing this! Have been testing and everything is running smooth except having issues with DHCP Reservations and clients not getting that IP address.

gnuton, 2022-08-31:
@cjnickisherxcii unlikely that's a firmware issue

apjashley1, 2022-08-31:
Strange on refreshing the Guest Network page, there's options to make a Facebook network that disappear within seconds (the option itself is a button leading to a 404).

Also noticing a typo in the logs "waitting".

DrDagless, 2022-09-05:
Absolute immense work, thanks so much. Just got the mesh 2-pack and can't wait for the stable.

gnuton, 2022-09-06:
New Beta1 released, please upgrade. I feel like it's pretty safe to be tested for curious people too. 
Happy testing and Enjoy.

popadiys, 2022-09-06:
openvpn server, pppoe, ipv6 work as a charm. thanks!

apjashley1, 2022-09-06:
Good work getting the Beta out!

Found another typo, on the WiFi Radar page "Associated Station's" should be "Associated Stations" with no apostrophe

Hyphaed, 2022-09-07:
Could you add WireGuard VPN client please :: I do not see the option with the beta firmware

valuxin, 2022-09-09:
> Could you add WireGuard VPN client please :: I do not see the option with the beta firmware

Wireguard VPN WebGUI will be in 388 version. But you can already install it using amtm command.

Niko2040, 2022-09-09:
I've got a problem with this firmware: after some time (usually ~12h) all wi-fi spots are turnings off. I faced this problem on old official firmware before rev 42238, but it appeared not so often. It has been fixed since rev 42238 after contacting asus tech support and router work perfectly on latest official firmware.

gnuton, 2022-09-10:
@Niko2040 does this problem go away resetting to factory the router? do you see anything wrong in the logs?

Niko2040, 2022-09-10:
> @Niko2040 does this problem go away resetting to factory the router? do you see anything wrong in the logs?

Problem goes away after reflashing to official firmware. 
Log before wireless turns off:
```Sep  9 12:04:23 kernel: Print axmacphyIfTbl
Sep  9 12:04:23 kernel: Print vasipregister
Sep  9 12:04:23 kernel: Print vasippchistory
Sep  9 12:04:23 kernel: Print smctable
Sep  9 12:04:23 kernel: Print smctinytable
Sep  9 12:04:23 kernel: Print BFM table
Sep  9 12:04:23 kernel: Print bfdIndexLUT
Sep  9 12:04:37 kernel: dhd_os_stop_wd_thread(): thread:dhd_watchdog_thread:2d8 terminated OK
Sep  9 12:04:37 kernel: CFG80211-ERROR) wl_add_remove_eventmsg : Get event_msgs error (-1)
Sep  9 12:04:37 kernel: br0: port 7(eth7) entered STP off state
Sep  9 12:04:37 kernel: device eth7 left promiscuous mode
Sep  9 12:04:37 kernel: br0: port 7(eth7) entered STP off state
Sep  9 12:04:37 kernel: cpu Ring 2 has been flushed
Sep  9 12:04:37 kernel: kthread_should_stop detected in wfd
Sep  9 12:04:38 kernel: dhd_module_init in
Sep  9 12:04:38 kernel: DHD_FKB_POOL size is:1280 and entry size:2816
Sep  9 12:04:38 kernel: DHD_PKTTAG POOL size is:2592 and entry size:128
Sep  9 12:04:38 kernel: dhd_queue_budget = 256
Sep  9 12:04:38 kernel: dhd_sta_threshold = 2048
Sep  9 12:04:38 kernel: dhd_if_threshold = 65536
Sep  9 12:04:38 kernel: no wifi platform data, skip
Sep  9 12:04:38 kernel: PCI_PROBE:  bus 1, slot 0,vendor 14E4, device AAA4(good PCI location)
Sep  9 12:04:38 kernel: dhdpcie_init: can't find adapter info for this chip
Sep  9 12:04:39 kernel: DHD: dongle ram size is set to 6291456(orig 6291456) at 0x0
Sep  9 12:04:39 kernel: dhd:0: fw path:/etc/wlan/dhd nv path:(null)
Sep  9 12:04:39 kernel: dor0: runner supported ring format types TxP 0x5, RxP 0x3 TxC 0x3 RxC 0x3^M
Sep  9 12:04:39 kernel: dhd_runner_attach: Tx Offload - Enabled, Ring Size = 1024
Sep  9 12:04:39 kernel: dhd_runner_attach: Rx Offload - Enabled, Ring Size = 1024
Sep  9 12:04:39 kernel: TX wakeup info: reg = <0x82299004>, val = <0x10000000>
Sep  9 12:04:39 kernel: RX wakeup info: reg = <0x8229a004>, val = <0x28000000>
Sep  9 12:04:39 kernel: CFG80211-ERROR) wl_setup_wiphy : Registering Vendor80211
Sep  9 12:04:39 kernel: dhd_attach(): thread:dhd_watchdog_thread:3df1 started
Sep  9 12:04:39 kernel: dhd_deferred_work_init: work queue initialized 
Sep  9 12:04:39 kernel: Creating CPU ring for queue number 4 with 1024 packets descriptor=0xffffffbffc54b9c0
Sep  9 12:04:39 kernel:  Creating CPU ring for queue number 5 with 1024 packets descriptor=0xffffffbffc54ba00
Sep  9 12:04:39 kernel:  ^[[1m^[[34m wfd_bind: Dev eth%d wfd_idx 2 wl_radio_idx 0 Type fkb configured WFD thread wfd2-thrd minQId/maxQId (12/13), status (0) qmask 0x30^[[0m
Sep  9 12:04:39 kernel: Instantiating WFD 2 thread
Sep  9 12:04:39 kernel: dhdpcie_irq_enabled: depth:0
Sep  9 12:04:39 kernel: dhd:0: fw path:/etc/wlan/dhd nv path:(null)
Sep  9 12:04:39 kernel: DHD  concate_extfdimg_name: Final FW path is /etc/wlan/dhd/43684b0/rtecdc.bin 
Sep  9 12:04:39 kernel: dhd_bus_download_firmware: firmware path=/etc/wlan/dhd/43684b0/rtecdc.bin, nvram path=
Sep  9 12:04:39 kernel: dhdpcie_ramsize_adj: Enter
Sep  9 12:04:39 kernel: dhd_bea_read_header: Not a .bea header
Sep  9 12:04:39 kernel: dhdpcie_ramsize_adj: Adjust dongle RAMSIZE to 0x580000
Sep  9 12:04:39 kernel: dhdpcie_download_code_file: download firmware /etc/wlan/dhd/43684b0/rtecdc.bin
Sep  9 12:04:39 kernel: dhd_bea_read_header: Not a .bea header
Sep  9 12:04:39 kernel: dhdpcie_bus_write_vars: Download, Upload and compare of NVRAM succeeded.
Sep  9 12:04:39 kernel: PCIe IPC addr (0x0017800c) read took 43023 usec
Sep  9 12:04:39 kernel: PCIe IPC: addr 0x0017800c sz 128, Rev host 0x82 dngl 0x82
Sep  9 12:04:39 kernel: 		FWID 0x271c1695, flags 0x10b10082, dcap1 0x2f8e0000 dcap2 0x00000001
Sep  9 12:04:39 kernel: 		trap 0x00000000 assrt 0x00000000,0x00000000,0 cons 0x0024d904
Sep  9 12:04:39 kernel: 		mbox h2d 0x00275ef4 d2h 0x00275ef8 rings 0x00273494
Sep  9 12:04:39 kernel: 		HME Pages: 1 0 0 0 0 32 0 0
Sep  9 12:04:39 kernel: PCIe IPC RINGS sz 128: ring_mem 0x00273514
Sep  9 12:04:39 kernel: 		h2d wr 0x00275694 rd 0x00275ab8 d2h wr 0x00275edc rd 0x00275ee8
Sep  9 12:04:39 kernel: 		rings h2d 530 d2h 6 flows 528 ifs 15
Sep  9 12:04:39 kernel: PCIe IPC: d2h_sync_mode 0x00000000
Sep  9 12:04:39 kernel: PCIe IPC: Max rings: h2d 530 d2h 6
Sep  9 12:04:39 kernel: PCIe IPC: 2 bytes sized indices
Sep  9 12:04:39 kernel: Host iDMA enable
Sep  9 12:04:39 kernel: PCIe IPC: DMAing indices: H2D 1 D2H 1
Sep  9 12:04:39 kernel: H2D DMA WR INDX : ffffff8004799000 array size 1088 = 2 * 530
Sep  9 12:04:39 kernel: D2H DMA RD INDX : ffffff8004799800 array size 64 = 2 * 6
Sep  9 12:04:39 kernel: D2H DMA WR INDX : ffffff8004799c00 array size 64 = 2 * 6
Sep  9 12:04:39 kernel: H2D DMA RD INDX : ffffff800479a000 array size 1088 = 2 * 530
Sep  9 12:04:39 kernel: PCIe IPC: DHD HDR 1
[... 308 lines cut ...]
Sep  9 12:04:43 kernel: CONSOLE: 097777.060 wl0: wlc_bmac_wowlucode_start mctrl=0x4020402 0x4020402
Sep  9 12:04:43 kernel: CONSOLE: 097777.062 wl0: CORE INIT : mode 4 pktclassify 1 rxsplit 1  hdr conversion 1 DMA_CT Enabled
Sep  9 12:04:43 kernel: CONSOLE: 097777.084 wlc_mutx_active_update vasip mu_supported_Ntx 4
Sep  9 12:04:43 kernel: CONSOLE: 097777.084 HWA1a RxPOST: H2D RxPost ring: id<1> type<0> item_type<1> max_items<1024> len_item<8>
[... cut ...]
Sep  9 12:04:46 kernel: CONSOLE: 097779.829 wl0: wlc_iovar_ext: chanspec: BCME -20
Sep  9 12:04:46 kernel: br0: topology change detected, propagating
Sep  9 12:04:46 kernel: br0: port 5(eth5) entered forwarding state
Sep  9 12:04:47 kernel: br0: topology change detected, propagating
Sep  9 12:04:47 kernel: br0: port 6(eth6) entered forwarding state
Sep  9 12:04:47 kernel: br0: topology change detected, propagating
Sep  9 12:04:47 kernel: br0: port 7(eth7) entered forwarding state
Sep  9 12:04:54 roamast: ROAMING Start...```

gnuton, 2022-09-11:
@Niko2040 
cannot see anything wrong in the logs. What time do you loose the connection?
Have you reset to factory after flashing Merlin? if so do you still see the problem?

gnuton, 2022-09-12:
Beta2 released with wireguard support

apjashley1, 2022-09-12:
> 

Is it in the GUI? I don't see it under VPN

apjashley1, 2022-09-12:
On the AImesh page for the mesh node only, the SSID for Wireless - 5 GHz-2 is reported as the default (ASUS_XX_5G-2_Guest5) not the actual SSID being broadcast

gnuton, 2022-09-13:
@apjashley1 no GUI for wireguard. I guess you can use this https://github.com/MartineauUK/wireguard or no idea if there is anything better since  I do not use it myself.

high1, 2022-09-13:
Beta has been running strong here, without any issues here since the day it was released. Thanks for this!

apjashley1, 2022-09-13:
> is that 5 GHz-2 network for aimesh router2router communication? I have not checked ont his model but on Zen it works like that.

No, it's the same 5GHz-2 network as the main router on Smart Connect (2 hardwired AX92u) but when you click the mesh node this SSID doesn't match the actual SSID being broadcast from it.

gnuton, 2022-09-13:
@apjashley1 have you seen this issue on stock firmware?

apjashley1, 2022-09-14:
@gnuton didn't see it on stock firmware, not sure about alpha or beta1 (sorry!)

Niko2040, 2022-09-14:
> What time do you loose the connection?
Usually within 12 hours
> Have you reset to factory after flashing Merlin?
Yes. Every time I flash new firmware (even official), I do factory reset to prevent bugs with old settings.

Now I flashed beta2 and didn't turn on QoS, and the router has been working for more than 1 day without problem.

gnuton, 2022-09-14:
@Niko2040 so the problem seems to be in the QoS config. Try to switch it on again and see you get back the disconnections

yahelgr, 2022-09-25:
When will the stable version be released?

cauddy, 2022-09-27:
Looks like the option to enable 5g 160mhz band has disappeared?
The official asus router firmware had this option.

gnuton, 2022-09-27:
@cauddy which stock firmware version? I will double check if it s my fault..

gnuton, 2022-09-27:
@yahelgr this or next weekend. I just need to find some time

cauddy, 2022-09-27:
> @cauddy which stock firmware version? I will double check if it s my fault..

3.0.0.4.386.46061 definitely had the DFS Channels for 5g-2.

I remember the WebUI might not have had the 160mhz band for 5g-2, but was settable via the Asus Router Android App. Appears that using this new MerlinWRT Fork firmware, has also had a negative effect on Asus Router App, where using the app if you go into Wireless Settings it has no options available for the 5g Wireless except for turning on and off the radio. All other options have disappeared.

otaking71, 2022-09-27:
Ive been using your forks for a while.   This firmware definitely makes the ax92u on helluva mesh unit.    I just have one issue that seems to only work on your fork of the xt8

The secondary and tertiary guest networks dont show up on the nodes.   The xt8 running your fork did it just fine.   They were controlled by the primary.   Ive not tested on the stock xt8...but the ax82 and et8 do not behave the way the forked xt8 did.

valuxin, 2022-09-28:
> Looks like the option to enable 5g 160mhz band has disappeared? Also no option to enable DFS channels on 5g-2?
> 
> The official asus router firmware had this option.

I'm on Beta 1 firmware and don't have such problems. I have this options in WebUI and Asus Router App works the same as before.

kaluchka, 2022-09-28:
Running the latest official fw version, clients were disconnecting form the guest network after 10 minutes a and were unable to reconnect again. CS of Asus wasn't able to help me and suggested to return the AX92U which I bought 2 weeks ago to Amazon. Before doing so I tried the Beta 2 and everything is working fine!!! No more disconnections from the guest network. The only thing I am missing is a Wireguard client VPN. Thank you very much!

gnuton, 2022-09-29:
@kaluchka wireguard is part of beta2. I think you can use https://github.com/MartineauUK/wireguard to set it up

rafaqueque, 2022-09-29:
I've been using Beta 2 in my main router and no issues found so far. Thank you!

kaluchka, 2022-09-29:
> @kaluchka wireguard is part of beta2. There is no UI although, so I think you should use https://github.com/MartineauUK/wireguard to set it up. Honestly I have not tested it

Unfortunately I am not really acquainted with the subject of the page you referred to. For me it is like Chinese :) Still excellent work, my router is running without any flow. Thanks again

DrDagless, 2022-09-29:
I've been using the beta 1 since it launched and it's largely been great with my main + node. I have had two times where internet dropped and I had to restart the main router (which fixed it). Happy to send a log over if that'd help.

nichtsnutz0815, 2022-09-29:
I have installed the beta 1 and I´m happy with it. No problems till now. great work. thanks a lot.

PacificTrade, 2022-10-02:
Will get this installed on my mesh nodes tonight. Big thanks for working on this!

gnuton, 2022-10-08:
Stable release is now available.
For tracking issues from now on, please open an ad-hok ticket.
Thanks

closed 2022-10-08 by gnuton

curtisling, 2022-12-03:
Do you think it might be possible to do a build for the AX92U of 388.1 (3-Dec-2022)?

[1 more reply]