gnuton/asuswrt-merlin.ng #237: LED management setting for XT8 not working

derek286, 2022-08-16:
**Router Model Affected**
Models: XT8

**Firmware Version Affected**
386.07_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
The LED for all nodes can be turned off under the AiMesh Management. But whenever the node is rebooted manually or all the nodes are rebooted by the scheduler, the LED would be turned on no mater the setting is shown as Off. It can only be turned off again by toggle the setting, i.e. change to On and then Off. 

**To Reproduce**
Steps to reproduce the behavior:

1. Go to 'AiMesh'
2. Click any XT8 Node 
3. Click on 'Management'
4. Click on the setting for 'LED' to turn it Off
5. Reboot the Node and will find the LED is On
6. To turn it off, need to repeat the above step 4 to turn it to On and then Off

**Expected behavior**
The LED should be Off and any reboot should not affect the setting.

**Screenshots**
The LED was On when this was set to Off.
[image]

gnuton, 2022-08-16:
do you know if it was working before?

derek286, 2022-08-17:
> 

Yes this has been working fine with the stock released and beta firmware from ASUS. I have just switched to use Merlin for a week.

gnuton, 2022-08-22:
cool. my question was more if you tried my previous stable, but I think I got you didn't.
I will check this.
Thanks for reporting.

reerden91, 2022-10-16:
Can confirm this issue on AX-92U with  386.08-gnuton1 as well.

gnuton, 2023-12-10:
linked to https://github.com/gnuton/asuswrt-merlin.ng/issues/506
and https://github.com/gnuton/asuswrt-merlin.ng/issues/489
fix https://github.com/gnuton/asuswrt-merlin.ng/commit/0281409b0b9ef560a845baee771781021f83e9af