gnuton/asuswrt-merlin.ng #240: Use full upload speeds blocks the network to the Main Router and Internet

MrRojano97, 2022-08-23:
**Router Model Affected**
Models: TUF-AX5400

**Firmware Version Affected**
386.07_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
I don't know

**Describe the bug**
If a device uses the full of the network upload speed (in my case I have 940 Mbps of upload speed from my ISP), the entire network will lose the internet connection and the connection to the main router at the time of the upload. I can replicate the issue forcing the QoS to be 500 Mbps too, so the problem is strange because the cause is not the ISP speed.

Furthermore, while the upload test is running I can't do pings to the main router (192.168.50.1) but yes to others TUF-AX5400 (with the same firmware) connected via backhaul, even I can ping the other TUF-AX5400 node. The topologies I tested was:

1. Computer -> Ethernet -> TUF-AX5400 Node (connected via EthBackhaul) -> Ethernet -> TUF-AX5400 (Main Router) -> PPPoE ISP
2. Computer -> Ethernet -> TUF-AX5400 (Main Router) -> PPPoE ISP

I also tried different Ethernet adapters such as Realtek 2.5 GbE, 1 GbE and ASIX 1 GbE.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Speedtest.net and perform a Speedtest using an Ethernet 1 GbE Adapter (or 2.5)
2. While the Upload test is running, try to perform a ping on Terminal or CMD to 192.168.50.1 or google.com
3. The pings will result on a timeout error, then, when the upload test ends will resume the connection and the ping will work.

**Expected behavior**
The ping and the average stability of the network would be unaffected by performing an upload test or an upload of a large amount of files using the full upload speed.

**More interesting info**
If you have a SAMBA device on your network, you can upload files to him using the full Gigabit upload speed and the ping and network will continue working without problems. In fact, I tried the next topology and the network continues entirely stable.

Computer Uploading data to SAMBA -> Ethernet -> TUF-AX5400 Node (Connected via EthBackhaul) -> Ethernet -> TUF-AX5400 (Main Router) -> Ethernet -> Computer with Samba receiving data

**Screenshots**
[image]
[image]

gnuton, 2022-08-29:
Hi,
If I got it right, you have computer A -> router -> internet
and you browse speedtest.net on computer A.
Then you run ping on computer A and the ping doesnt' work. Am i right?

trebleta, 2022-08-29:
I've noticed something like this, so checked on offical, on my asus dsl ax82u. Here the post
https://rog.asus.com/forum/showthread.php?130249-WiFi-issule-DSL-AX82U-AX5400.
For me tho its just the asus router that I have problems with, other devices still connect and work fine untill i try to access the asus device.

trebleta, 2022-08-29:
In QOS I am using bandwith limiter for some but not all devices. It's the devices not in the list using too much overhead.

MrRojano97, 2022-10-01:
Hi, tested disabling all Asus additional stuff except AiMesh and works perfectly fine. (AiProtect, Router App Control, Traffic Analyzer, etc)