gnuton/asuswrt-merlin.ng #241: MAP-E not working properly

wspruce, 2022-08-24:
**Router Model Affected**
Models: AX-5400

**Firmware Version Affected**
386.07

**Is this bug present in upstream Merlin releases too?**
unknown

**Describe the bug**
I use the v6Plus(IPv4overIPv6) mode with AX-5400 in Japan. When using official fw [Version 3.0.0.4.386.49873](https://www.asus.com/Networking-IoT-Servers/WiFi-Routers/ASUS-Gaming-Routers/TUF-Gaming-AX5400/HelpDesk_BIOS/) and before, the logs I get are:
```
May  5 14:05:52 [MAP-E]: v6plus starts.
May  5 14:06:24 [MAP-E]: Receiving a rule is successfully(200). Reconfirm after 1079 min.
```
It uses 32s to receive a map rule.

but when using TUF-AX5400_386.07_2-gnuton1 I get
```
May  5 14:05:57 [MAP-E]: v6plus starts.
May  5 14:06:27 [MAP-E]: Receiveing invalid rule. Reconfirm after 20 min.
```
It looks like a timeout after 30s.

**To Reproduce**
I don't know. Maybe move to Japan and get a v6plus ISP? 😓

---
Also I noticed that in the release note of official fw Version 3.0.0.4.386.43588 it said:
> ASUS TUF-AX5400 Firmware version 3.0.0.4.386.43588
> **1. Add JPNE v6plus support.**
> ...

May this be the hint?

gnuton, 2022-08-24:
Have you tested an earlier version of the stock firmware? If so which one? Did it work?

gnuton, 2022-08-24:
If you wanna keep Merlin try to downgrade to the previous stable and let me know if it works

wspruce, 2022-08-25:
At least Version 3.0.0.4.386.46061 and Version 3.0.0.4.386.48664 of the stock firmware also worked. Didn't try anything before them.

And the previous stable ([386.5_2-gnuton0](https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/386.5_2-gnuton0)) don't work either with same error:
```
Aug 25 22:05:33 [MAP-E]: Receiveing invalid rule. Reconfirm after 28 min.
```