gnuton/asuswrt-merlin.ng #242: auto updates like stock firmware

lsaadeh, 2022-08-25:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

it would be useful to have the auto-update feature like in the stock firmware.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

implementation of auto-update feature like in the stock firmware.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

https://www.asuslogin.com/how-to-enable-and-schedule-the-automatic-firmware-update-for-asus-router/

DervilRus, 2022-08-25:
Merlin wrote on the forum that for security reasons he will not implement this feature. Because This firmware is a fork of Merlin, I think it will not be implemented here either.

gnuton, 2022-08-26:
I could enable it, but I believe this is a feature common to all routers, so it would be nice to get that implemented upstream.