gnuton/asuswrt-merlin.ng #243: DoT makes the router unstable - breaks the internet in 5Ghz band

peros550, 2022-08-30:
**Router Model Affected**
Models: DSL-AC68U

**Firmware Version Affected**
386.7_2gnuton1

**Is this bug present in upstream Merlin releases too?**
No, I use DoT without issues in RT-AC68U.

**Describe the bug**
When enabling DoT, Internet stops working although the WAN seems connected. Eventually, after quite some time, it may become available again only for 2.4Ghz band. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to WAN page. Edit the interface, enable DoT. 

**Expected behavior**
Internet should continue working without. 

**Screenshots**
Sorry can't provide screenshots. However, if nobody else can reproduce the issue I will have to upload videos. Let me know.

gnuton, 2022-08-31:
can you see anything wrong in the logs?

peros550, 2022-08-31:
unfortunately nothing related to WAN connection or DOT

trebleta, 2022-09-02:
Can you link your DOT setting so I can check there correct

peros550, 2022-09-05:
My bad here. It turned out, I enabled the DOT but didn't set any server. Maybe some room for improvement, would be to replicate settings as in the RT-AC68 running Merlin, in which there is a list of preconfigured servers to select.

closed 2022-09-05 by peros550