gnuton/asuswrt-merlin.ng #244: DSL-AX82U Freeze completely firmware

dimvas243, 2022-09-01:
**Router Model Affected**
Models: DSL-AX82U

**Firmware Version Affected**
386.07_2-gnuton1 - 386.08_0-gnuton0_beta1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
In both firmware the router freezes complete no internet, no wifi, no lan with out changing any settings from firmware 386.5_2-gnuton0 help me please to debug it i don't know how to get the logs after freeze so i can help you further.

thank you in advance

gnuton, 2022-09-01:
have you reset the router to factory?

dimvas243, 2022-09-01:
thanks for the response i didn't try it  one small question, can i do factory reset and restore setting from backup or i must restore them manually

Thank you in advance

trebleta, 2022-09-02:
Do a hard reset, by turning off the router holding the wps button switching on then wait till the led changes the switch of and on again.
Yes manualy do all setting, to rule that out

gnuton, 2022-09-05:
@dimvas243 @trebleta is right do not use backups. Merlin is similar but not the same as stock firmware.

dimvas243, 2022-09-10:
i have hard reseted, manual reconfigure evrething stil freezes red light on dsl, no  ssh connection with the message  "kex_exchange_identification: read: Connection reset", the page will not open. The freeze happend after 1 - 2 hours i have install flex qos

trebleta, 2022-09-11:
And if flex QOS is not installed?

dimvas243, 2022-09-12:
without flexqos running 24+ hours straight without any problem

dimvas243, 2022-09-13:
i have tested with the developement repo and no freeze for 24 hours i am closing it

closed 2022-09-13 by dimvas243