gnuton/asuswrt-merlin.ng #245: WAN Port - ISP DHCP Data Fetching

GWraith1985, 2022-09-05:
_### **Router Model Affected**_
Models: ZenWiFi_XT8 (base model: RT-AX95Q)   **Never occurred on my older model AX92U!**

_### **Firmware Version Affected**_
386.07_2-gnuton1

_### **Is this bug present in upstream Merlin releases too?**_
Yes - Was also present in regular ASUS stock firmware

_### **Describe the bug**_
Anytime you reset or restart the router you are met with a message that your ISP DCHP doesn't function properly. It's immediately fixed by unplugging the WAN ethernet from the router and plugging it back in.

_### **To Reproduce**_
Steps to reproduce the behavior:
1. Reboot Router
2. Check Internet Connected Status on Main Page to verify the error message
3.  Unplug the WAN cable, wait 2 seconds
4. Plug the WAN cable back in, and the issue is immediately resolved.

_### **Expected behavior**_
This should occur without intervention automatically. However, my investigation leads me to believe that this might have something to do specifically with FRontire FIOS fiber internet through the FOX222 ONT. Could possibly add a line of start-up code to reset the WAN port to emulate a unplug or add a button for the user to do so without going to the router. I have tried suggested changes to the [DHCP query frequency] setting and none of the three available solve the issue.

_### **Screenshots**_
I don't believe this is needed but I can provide it if necessary. To date, ASUS themselves has not been unable to solve this issue.

**P.S.
Please also see this post of issues with ASUS firmware not allowing setting DNS to the same IPs without doing it through NVRAM. This only affects PiHole users for reporting and logging spam. 

https://www.reddit.com/r/pihole/comments/u9ecwd/asus_ax92u_pihole/**

gnuton, 2022-09-05:
not sure yet how to help you. 
do you have any log?
Mixing AX92U with XT8 puzzles me. 
Adding pyhole in the end another issue at the complicates even more the things..

GWraith1985, 2022-09-10:
I will capture the logs this evening.

The only hardware I am using is the ASUS XT8 and a pinhole on a Pi4B.

Two issues exist with stock and your fork that while not software breaking are a nuisance in my environment.

1. XT8 / FOX222 ONT (10Gbe Port)
   A. Router will not automatically poll and pull WAN IP from frontier servers upon restart. You must go and unplug the WAN cable from the router and plug it back in once the router is fully booted.

2. ASUS Firmware doesn't natively allow two of the same DNS IPs.
   B. Even though it shows you that you put in two DNS entries the same it will ignore the second one. Specifically home based in environments where redundancy is not common you then have logs on your DNS server which will only show the router IP as sending all the DNS queries. I linked my article above from my original router when I figured out how to fix it. 

For either of these, I was not sure if you could figure out the issue or code in the solution. I have contacted ASUS a few times and submitted logs and fixes and they never respond or fix the issue.

trebleta, 2022-09-11:
It's a common issule on asus, apon starting it  say our ISP DCHP doesn't function properly, as the router is still booting it self. If after 10-20 min and its still not then google asus forum, hardware-software support post there.

GWraith1985, 2022-10-05:
Oct  5 07:08:51 kernel: eth1 (Ext switch port: 0) (Logical Port: 8) (phyId: 0) Link DOWN.
Oct  5 07:09:01 kernel: eth1 (Ext switch port: 0) (Logical Port: 8) (phyId: 0) Link Up at 100 mbps full duplex
Oct  5 19:18:05 Mastiff: Got SIGTERM
May  5 01:05:20 kernel: brcmboard registered
May  5 01:05:20 kernel: broadcomThermalDrv brcm-therm: init (CPU count 4 4 4 4)
May  5 01:05:21 kernel: eth1 (Ext switch port: 0) (Logical Port: 8) (phyId: 0) Link Up at 100 mbps full duplex
May  5 01:05:22 kernel: eth3 (Ext switch port: 2) (Logical Port: 10) (phyId: 2) Link Up at 100 mbps full duplex
May  5 01:05:23 acsd: eth4: Selecting 2g band ACS policy
May  5 01:05:23 ZenWiFi_XT8: start httpd:80
May  5 01:05:23 httpd: Save SSL certificate...80
May  5 01:05:23 httpd: mssl_cert_key_match : PASS
May  5 01:05:23 disk monitor: be idle
May  5 01:05:23 httpd: Succeed to init SSL certificate...80
May  5 01:05:23 hour monitor: daemon is starting
May  5 01:05:23 hour monitor: daemon terminates
May  5 01:05:23 kernel: eth2 (Ext switch port: 1) (Logical Port: 9) (phyId: 1) Link Up at 1000 mbps full duplex
May  5 01:05:23 dbus: daemon is started
May  5 01:05:24 jffs2: valid logs(1)
May  5 01:05:24 Mastiff: init
May  5 01:05:24 ERP: The model isn't under EU SKU!
May  5 01:05:26 WAN Connection: WAN(0) link down.
May  5 01:05:26 acsd: eth4: selected channel spec: 0x1006 (6)
May  5 01:05:26 acsd: eth4: Adjusted channel spec: 0x1006 (6)
May  5 01:05:26 acsd: eth4: selected channel spec: 0x1006 (6)
May  5 01:05:26 acsd: acs_set_chspec: 0x1006 (6) for reason APCS_INIT
May  5 01:05:26 acsd: eth5: Selecting 5g band ACS policy
May  5 01:05:27 kernel: eth0 (Int switch port: 1) (Logical Port: 1) (phyId: 7) Link Up at 2500 mbps full duplex
May  5 01:05:27 ahs: [read_json]Update ahs JSON file.
May  5 01:05:28 acsd: eth5: selected channel spec: 0xe22a (44/80)
May  5 01:05:28 acsd: eth5: Adjusted channel spec: 0xe22a (44/80)
May  5 01:05:28 acsd: eth5: selected channel spec: 0xe22a (44/80)
May  5 01:05:28 acsd: acs_set_chspec: 0xe22a (44/80) for reason APCS_INIT
May  5 01:05:28 acsd: eth6: Selecting 5g band ACS policy
May  5 01:05:29 acsd: eth6: selected channel spec: 0xe09b (149/80)
May  5 01:05:29 acsd: eth6: Adjusted channel spec: 0xe09b (149/80)
May  5 01:05:29 acsd: eth6: selected channel spec: 0xe09b (149/80)
May  5 01:05:29 acsd: acs_set_chspec: 0xe09b (149/80) for reason APCS_INIT
May  5 01:05:33 roamast: ROAMING Start...
May  5 01:05:44 kernel: eth1 (Ext switch port: 0) (Logical Port: 8) (phyId: 0) Link DOWN.
May  5 01:05:46 kernel: eth1 (Ext switch port: 0) (Logical Port: 8) (phyId: 0) Link Up at 100 mbps full duplex
May  5 01:06:24 rc_service: conn_diag 1793:notify_rc restart_amas_portstatus
May  5 01:06:46 rc_service: httpd 1714:notify_rc restart_wan_if 0
May  5 01:06:50 kernel: eth1 (Ext switch port: 0) (Logical Port: 8) (phyId: 0) Link DOWN.
May  5 01:06:52 kernel: eth1 (Ext switch port: 0) (Logical Port: 8) (phyId: 0) Link Up at 100 mbps full duplex
May  5 01:06:56 rc_service: httpd 1714:notify_rc restart_wan_if 0
May  5 01:07:26 kernel: eth0 (Int switch port: 1) (Logical Port: 1) (phyId: 7) Link DOWN.
May  5 01:07:30 kernel: eth0 (Int switch port: 1) (Logical Port: 1) (phyId: 7) Link Up at 2500 mbps full duplex
May  5 01:07:34 WAN Connection: WAN(0) link up.
May  5 01:07:34 rc_service: wanduck 1600:notify_rc restart_wan_if 0
May  5 01:07:38 wan: finish adding multi routes
May  5 01:07:39 WAN Connection: WAN was restored.
May  5 01:07:40 rc_service: udhcpc_wan 2857:notify_rc stop_samba
May  5 01:07:40 rc_service: udhcpc_wan 2857:notify_rc start_samba
May  5 01:07:40 rc_service: waitting "stop_samba" via udhcpc_wan ...
May  5 01:07:40 Samba Server: smb daemon is stopped

 (Unplugged and replugged WAN cable issue corrected.)

May  5 01:07:41 dhcp client: bound XX.XXX.XX.XX/255.255.255.0 via XX.XX.XX.1 for 1800 seconds.
May  5 01:07:41 ntp: start NTP update
Oct  5 19:22:05 rc_service: ntp 2958:notify_rc restart_diskmon
Oct  5 19:22:05 disk_monitor: Finish
Oct  5 19:22:05 disk monitor: be idle
Oct  5 19:22:10 rc_service: service 3305:notify_rc restart_firewall
Oct  5 19:23:47 WATCHDOG: [FAUPGRADE][auto_firmware_check:(7708)]periodic_check AM 4:54
Oct  5 19:23:47 WATCHDOG: [FAUPGRADE][auto_firmware_check:(7747)]do webs_update
Oct  5 19:23:57 WATCHDOG: [FAUPGRADE][auto_firmware_check:(7765)]retrieve firmware information
Oct  5 19:23:57 WATCHDOG: [FAUPGRADE][auto_firmware_check:(7768)]user in use
Oct  5 19:24:02 ahs: [read_json]Update ahs JSON file.
Oct  5 19:24:26 WATCHDOG: [FAUPGRADE][auto_firmware_check:(7747)]do webs_update
Oct  5 19:24:26 WATCHDOG: [FAUPGRADE][auto_firmware_check:(7765)]retrieve firmware information
Oct  5 19:24:26 WATCHDOG: [FAUPGRADE][auto_firmware_check:(7768)]user in use
Oct  5 19:24:56 WATCHDOG: [FAUPGRADE][auto_firmware_check:(7747)]do webs_update
Oct  5 19:24:56 WATCHDOG: [FAUPGRADE][auto_firmware_check:(7765)]retrieve firmware information
Oct  5 19:24:56 WATCHDOG: [FAUPGRADE][auto_firmware_check:(7768)]user in use

GWraith1985, 2022-10-05:
This was an upgrade to the latest ASUS version available which then would not pull WAN IP till I replugged the WAN cable. Soft turning off WAN and back on did not fix it.

tynril, 2025-01-10:
I have the same issue with the same ISP. I'm currently running `3004.388.8_2-gnuton1` on a ZenWifi XT8 (base model `RT-AX95Q`), connected to a Frontier FiOS ONT (`XGS-PON ONT`, model `FOX222`).

On start, the router fails to establish WAN connection (admin UI shows "problem with your ISP DHCP"). Unplugging the WAN cable briefly solves the issue, until the next router restart. Is there any kind of log or information I could provide that may help identify a root cause?

GWraith1985, 2025-01-11:
I had Frontier come out and replace the ONT with a new one different model. Was good after that.Sent from my iPhone