gnuton/asuswrt-merlin.ng #247: DSL-AX82U FAILING TO CONNECT TO ISP

Wayne7497, 2022-09-09:
**Router : DSL-AX82U/RT-
Version Affected** 386.07_2-gnuton1 & 386.08_0-gnuton0_beta1

386.07_2-gnuton1 would sync with ISP (ABB) but fail to retrieve DHCP.

Reverted back to 386.5_2-gnuton0 and its working fine

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
386.07_2-gnuton1 and 386.08_0-gnuton0_beta1 would sync with ISP (ABB) but fail to retrieve DHCP.

Reverted back to 386.5_2-gnuton0 and its working fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section

trebleta, 2022-09-11:
You say unable to get a dhcp, how long did you give, can take 5-10 min. This was a problem on offical release but was ment to be fixed,

Wayne7497, 2022-09-11:
I also have the optus DSL-AX5400 and it's only merlin's gnuton's 386.5_2 that's working for me on Optus's NBN. Gnuton's beta 386.08 still doesn't want to authenticate the DSL connection.

Btw, you can restore the DSL-AX82U features on the DSL-AX5400 (assuming you already have Gnuton's / Merlin's FW):

1) Login via Putty
2) nvram set odmpid
3) nvram set rc_support="dsl vdsl spectrum mssid 2.4G 5G usbX1 update switchctrl manual_stb 11AX pwrctrl nandflash smart_connect wifi2017 app ofdma wpa3 defpsk reboot_schedule ipv6 ipv6pt PARENTAL2 dnsfilter am_addons cake ntpd dnspriv dualwan mtwancfg pptpd openvpnd utf8_ssid printer modem webdav rrsut cloudsync media appbase timemachine hdspindown diskutility dnssec email bwdpi wrs_wbl ookla HTTPS letsencrypt ssh vpnc wl6 user_low_rssi tcode usericon jffs2 cfg_wps_btn stainfo cloudcheck realip ifttt alexa ipsec_srv mumimo netool cfg_sync no_finiwl amas bcmwifi bcmhnd mbo conndiag eula proxysta iperf3 mswan account_binding gameMode ftp_ssl hnd_ax_675x dhdlog dis11b"
4) nvram commit

I had both the DSL-AX82U (uk version) and the DSL-AX5400 so I did a comparison between the 2 nvram's to restore some of the lost features. The DSL-AX5400 had the odmpid=DSL-AX5400 whereas the DSL-AX82U was blank. In addition, the AX5400 had "noupdate" in the rc_support whereas the DSL-AX82U had "update" along with "bwdpi" which changes the Traffic Manager tab to AIProtection when you log in

If you don't want to keep entering these commands in everytime the router reboots, then
1) login via winscp
2) navigate to /jffs/scripts/
3) create file "init-nat" (file should show as init-nat and not init-nat.sh)
4) enter the following into the file:

#!/bin/sh
nvram set odmpid
nvram set rc_support="dsl vdsl spectrum mssid 2.4G 5G usbX1 update switchctrl manual_stb 11AX pwrctrl nandflash smart_connect wifi2017 app ofdma wpa3 defpsk reboot_schedule ipv6 ipv6pt PARENTAL2 dnsfilter am_addons cake ntpd dnspriv dualwan mtwancfg pptpd openvpnd utf8_ssid printer modem webdav rrsut cloudsync media appbase timemachine hdspindown diskutility dnssec email bwdpi wrs_wbl ookla HTTPS letsencrypt ssh vpnc wl6 user_low_rssi tcode usericon jffs2 cfg_wps_btn stainfo cloudcheck realip ifttt alexa ipsec_srv mumimo netool cfg_sync no_finiwl amas bcmwifi bcmhnd mbo conndiag eula proxysta iperf3 mswan account_binding gameMode ftp_ssl hnd_ax_675x dhdlog dis11b"
nvram commit

5) save
6) open putty
7) run:
chmod a+rw /jffs/scripts/*

These modifications will remove the is_ax5400_i1 and is_ax5400_iln flags in isSupport() in the console.

I tried other startup scripts but the nvram kept getting overwritten during the startup and it was only the init-nat script that worked. When the router starts up, it'll show DSL-AX5400 for about a minute or two until the init-nat script is triggered, then the router will show "DSL-AX82U" with the restored lost features.

If anyone knows how to fix it more permanently, that would be much appreciated.

Wayne7497, 2022-09-17:
Thanks for all your efforts hopefully be fixed soon for us Australians

FreoPhil, 2022-10-01:
Same issue with AX82u in Australia with Aussie Broadband. Running previous version which is stable. I reported the same issue some time ago and it was dismissed. Glad you have reported the issue again.

gnuton, 2022-10-06:
duplicate of https://github.com/gnuton/asuswrt-merlin.ng/issues/241

closed 2022-10-06 by gnuton