gnuton/asuswrt-merlin.ng #248: TUF-AX5400 minor issues on 386.07_2-gnuton1

Rovasteen, 2022-09-10:
TUF-AX5400
386.07_2-gnuton1
Do not know if issue is on upstream Merlin

Issue 1:
AX devices will not connect on 5GHz band at 160 MHz using Wireless setting "20/40/80/160 MHz" and tick box "Enable 160 MHz".  If I change the Wireless setting to "160 MHz" and tick box "Enable 160 MHz" then my AX devices connect at 160 MHz..  Seems like an issue with setting "20/40/80/160 MHz" as this setting does not seem to allow 160 MHz channel connections.

Issue 2:
On the main menu General -> Tools, the Tools menu item has no icon.  Adding an icon would be a nice cosmetic item.  Not a big priority.  This is a minor issue.

gnuton, 2022-09-12:
Thanks for reporting. I will check this.

Rovasteen, 2022-09-14:
Wanted to thank you for your work in creating these builds.  Your work is appreciated.  The build TUF-AX5400
386.07_2-gnuton1 has been up now for a week on my router without any instability or crazy debug entries.  Well done Gnuton!

Rovasteen, 2022-10-06:
I tested the new Asus official build 3.0.0.4.386_50224.  On this build using Wireless setting "20/40/80/160 MHz" and tick box "Enable 160 MHz" plus selecting a specific channel (picked 36) allows 160 MHz to work.  Big difference between standard build and Gnuton IMHO.  On the official Asus builds the standard log functions are very basic, no tools, and lots of settings not included as compared to Merlin.  What is involved to create a new Gnuton build based on this later GPL?

gnuton, 2022-10-06:
@Rovasteen  We have already got  some 388 builds and the merge has already started. I think we will start to publish something in november.

bobyang3, 2022-12-29:
Ask a very dummy question. I am from AC68U, there is a "Reboot" button next to "Log Off", but I don't see it. How may I reboot the tuf-5400 router from the web UI?

gnuton, 2022-12-29:
@bobyang3 reboot button should be on the banner

gnuton, 2022-12-30:
@Rovasteen is  this bug still present in the latest 388 beta?

bobyang3, 2022-12-30:
> @bobyang3 reboot button should be on the banner

Do we have a different screen? I don't see it. Thanks

[image]

bobyang3, 2022-12-30:
@gnuton never mind, I found it. :)

[image]

Rovasteen, 2022-12-30:
Icon is there for tools on 388 Beta 2.