gnuton/asuswrt-merlin.ng #250: Flashed ax5400 to latest stable DSL-AX82U fw, but no extra features

dfgdgdgfd, 2022-10-01:
I have the merlin logo, options to use ssh etc, and ability to update firmware (This was isp-locked before the flash) but none of these:

Quick Internet Setup
TrendMicro stuff: AiProtection / Adaptive QoS / Traffic Analyzer / Bandwidth Monitor / Website history / Web & App Filter
Telnet / SSH
Router Feedback

Do I need to use the beta firmware to unlock these?  The main reason for flashing for me was to get rid of the isp-locked flash function.

gnuton, 2022-10-01:
Hi,
Thanks for flashing Merlin. The features are listed here https://www.asuswrt-merlin.net/features
All features you said are there. Some ISP blocked routers may have some page disabled or not visible but the funciontality is there.
I do not get BTW what's the problem since you say you have SSH but at same time you say ther is no SSH.

dfgdgdgfd, 2022-10-02:
My issue was that I didn't run these commands, I didn't see them in the instructions, are you sure they're there?

nvram set model=DSL-AX82U
nvram set productid=DSL-AX82U
nvram set odmpid=DSL-AX82U

nvram commit
reboot

gnuton, 2022-10-10:
I think I can fix this in the code.. let me give a try...

gnuton, 2022-10-10:
Can you please tell me what are odmpid productid and model by default in your router?
I see those are from the flash memory at every restart..

dfgdgdgfd, 2022-10-10:
Where do I find those?

gnuton, 2022-10-11:
ssh onto the router and run nvram get <var_name>

gnuton, 2022-10-11:
ssh to the router and run nvram get or something like that

viTRifY13, 2022-10-11:
Hi gnuton, for the DSL-AX5400, the default values are:
odmpid = DSL-AX5400
productid = DSL-AX82U
model = DSL-AX82U

What dfgdgdgfd is wanting is what I had originally posted (https://forums.whirlpool.net.au/thread/32k1vyn9?p=19) and is the following changes:
odmpid = 
rc_support = "dsl vdsl spectrum mssid 2.4G 5G usbX1 update switchctrl manual_stb 11AX pwrctrl nandflash smart_connect wifi2017 app ofdma wpa3 defpsk reboot_schedule ipv6 ipv6pt PARENTAL2 dnsfilter am_addons cake ntpd dnspriv dualwan mtwancfg pptpd openvpnd utf8_ssid printer modem webdav rrsut cloudsync media appbase timemachine hdspindown diskutility dnssec email bwdpi wrs_wbl ookla HTTPS letsencrypt ssh vpnc wl6 user_low_rssi tcode usericon jffs2 cfg_wps_btn stainfo cloudcheck realip ifttt alexa ipsec_srv mumimo netool cfg_sync no_finiwl amas bcmwifi bcmhnd mbo conndiag eula proxysta iperf3 mswan account_binding gameMode ftp_ssl hnd_ax_675x dhdlog dis11b"

What this does for the AX-5400 is remove the odmpid flag of DSL-AX5400 and unsets it to match the DSL-AX82U (dfgdgdgfd is incorrectly trying to set odmpid = DSL-AX82U whereas it needs to be unset).  In addition, it adds "bwdpi" and "update" to the rc_support and removes the "noupdate" flag

Thanks

reaps101, 2022-10-13:
@gnuton does the latest stable release 386.08_0-gnuton1 support tuf-ax5400 as i see in the notes there is a FIX: FlexQOS for TUF devices but no firmware for them thanks

gnuton, 2022-10-13:
@reaps101 no ther are no 386.08_0-gnuton1 images for TUF devices even if there are changes which actually affects all devices but that as secondary effect should fix an issue on TUF. https://github.com/gnuton/asuswrt-merlin.ng/issues/219

gnuton, 2022-10-13:
@reaps101 please report back your findings in the correct thread #219

basarbulut, 2022-10-22:
Hi. I am using 386.08_0-gnuton1 on my DSL-AX82U and relly happy with it. One thing I recognised is that the modem is forgetting internet speed results history in QoS page. (Adaptive QoS>Internet Speed>History). Any idea or something I am doing wrong?

gnuton, 2022-10-23:
Hi are you using flex?

basarbulut, 2022-10-23:
> Hi are you using flex?
> […](#)

I don't know what flex is.