gnuton/asuswrt-merlin.ng #251: Unable to connect to certain ISP

Stewdawg69, 2022-10-03:
**Router Model Affected**
DSL-AX82U

**Firmware Version Affected**
Latest Merlin versions (both Beta and Stable)

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
Unable to connect to specific ISP's. 

**To Reproduce**
This issue has been resolved in the latest release of the offical firmware from Asus (Version 3.0.0.4.386.49608).

The current stable release and this Beta release still has this issue present. 

I'm requesting that the next version of Merlin is based off the above offical firmware which has this issue resolved. 

Great work guys, 
Appreciate everything you guys do

gnuton, 2022-10-06:
Hi,
the latest beta was an attempt to fix this since it's based on 49634.
Just out of curiosity what's the name of the ISP and the country you have problem with?
I am defintely trying to fix this.. We are waiting for new GPLs from ASUS.

closed 2022-10-06 by gnuton

Stewdawg69, 2022-10-06:
> Currently with Aussie Broadband - Australia
>
> I checked the beta when posting the comment and it definitely didn’t work.
> Rolled back to the latest asus offical version and all working perfectly.
>
> Happy to be a Guinea pig for any releases as it does only affect some
> ISP’s, not all. So finding someone with the difficult ISP can he hard.
>
> Let me know if you need anything.
>
> Thanks,
> Stewart

I should also add after reading #241 (sorry I didn’t know the jargon when
posting) that Aussie Broadband is an IPV6 provider. I’m given both IPV6 and
IPV4 ip addresses.

Based on the thread I can only assume this appears to be related to the
issue.

Stewdawg69, 2022-10-11:
Currently with Aussie Broadband - Australia

I checked the beta when posting the comment and it definitely didn’t work.
Rolled back to the latest asus offical version and all working perfectly.

Happy to be a Guinea pig for any releases as it does only affect some
ISP’s, not all. So finding someone with the difficult ISP can he hard.

Let me know if you need anything.

Thanks,
Stewart

gnuton, 2022-10-11:
Thanks and please follow  #241 for updates.