gnuton/asuswrt-merlin.ng #254: Addon FlexQOS - Not appear in Menu - Section QOS

albertfxx, 2022-10-07:
**Router Model Affected**
Model: Tuf AX5400

**Firmware Version Affected**
386.07_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
Addon FlexQOS not appear in Menu-Section QOS of Merlin

**Additional info**
[dave14305 said:](https://www.snbforums.com/goto/post?id=780501)
FlexQoS expects to find the Classification tab (QoS_Stats.asp) from Merlin in the menu tabs, then insert its own tab right before it.

Since you don’t have that tab, you could try to update flexqos.sh to replace QoS_Stats.asp with AdaptiveQoS_WebHistory.asp on this line:

github.com
[FlexQoS/flexqos.sh at 2135fbd2c94d872b8cd825a0e96cbc5675fa2887 · dave14305/FlexQoS](https://github.com/dave14305/FlexQoS/blob/2135fbd2c94d872b8cd825a0e96cbc5675fa2887/flexqos.sh#L1248)
FlexQoS - Flexible QoS Enhancement Script for Adaptive QoS on ASUSWRT-Merlin - FlexQoS/flexqos.sh at 2135fbd2c94d872b8cd825a0e96cbc5675fa2887 · dave14305/FlexQoS
github.com

This is something [@GNUton](https://www.snbforums.com/members/63302/) would need to update in ej_tcclass_dump_array to make wan_iface = eth4 for this model.

gnuton, 2022-10-07:
This is a duplicate of https://github.com/gnuton/asuswrt-merlin.ng/issues/219
I have fixed this and it's in master but the current master doesn't build for TUF.
I will try to backport it and release an image for it. please follow  #219

closed 2022-10-07 by gnuton

albertfxx, 2022-10-08:
ok Sorry.
I will follow #219
Thank you!