gnuton/asuswrt-merlin.ng #255: Does the fw upgrade really available for this router?

F640, 2022-10-09:
**Router Model Affected**
Models: RT-AX82U

**Firmware Version Affected**
386.07_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
My router says it has update and changelog said "only available for DSL-AX82U and RT-AX92U"

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Administration
2. Click on Firmware Update
3. Check for update
4. Click on latest version (386.08-gnuton1)
5. It said something like in description

**Expected behavior**
No firmware upgrade for now like what changelog says

**Screenshots**
[image]

Camology, 2022-10-09:
Came to post the exact same thing.

gnuton, 2022-10-10:
Hi,
thanks for reporting.
You should not see anymore this notification.

closed 2022-10-10 by gnuton