gnuton/asuswrt-merlin.ng #256: XT8/RT-AX95Q Missing Ethernet Port Status

basementbuilder17, 2022-10-10:
**Router Model Affected**
Models: XT8/RT-AX95Q

**Firmware Version Affected**
386.07_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
Ethernet port status/checks are missing. Normally, when you view each of the Routers/Nodes (in the , it gives you the ethernet port status. Connected status (optimal=green/below speed = yellow), hover over it gives you the actual link speed (1 or 2.5 depending on port and capability), and it gave you basically like a status check. It seems that's all gone. I don't have any original firmware to post a screenshot, but it was there for the base firmware from factory. 

AIMesh Tab > Click on router or node > Network tab

This is where the ethernet port status/check used to be. Seems to be gone now.

d3v3l15h, 2023-01-28:
Same problem on 388.1_0-gnuton1. I just have the following lines in the UI (just this text and no icons) :

Ethernet Ports
2.5G
1
2
3
USB3.0
Connected
Warnings
Unplugged

rlfrank165, 2023-01-28:
A number of us have seen this before, and for reasons maybe Gnuton can
explain :-) try opening the UI in Incognito mode on your browser and
opening the UI from an incognito browsers window and see if this does
away...

Randy Frank

d3v3l15h, 2023-01-28:
Reflashed the XT8 before your answer and UI was correct then...

But whatever... I reverted back to 386. 388.1_0-gnuton1 is completely f... up on XT8, same as latest Asus firmwares. I have the exact same symptom than with Asus firmware : Aimesh node has very weak 2.4 ghz wireless connection (almost always disconnected in fact) while on 386 it's steadily connected on 5Ghz 2. As soon as I revert back both nodes on 386, everything is perfect again.

basementbuilder17, 2023-02-26:
So, I updated to 388. At first, i had the same "text only" feedback as see above. But today i logged in again, and it all seems to be working! I have the base GUI that shows all he ports with colors (full speed or limited) and hovering over gives me the actual speed. So from my point of view, this seems solved, unless it's an intermittent thing that comes back? Interesting to hear others feedback. Thanks @gnuton