gnuton/asuswrt-merlin.ng #259: DSL-AC68U GRE Tunnel

MHKalantarian, 2022-10-15:
**Is your feature request related to a problem? Please describe.**
I want to have GRE Tunnel feature on dsl-ac68u model

**Describe the solution you'd like**
I've looked around and people said adding CONFIG_NET_IPGRE=m to config file enables it, but I have 0 idea where the config file is or where to look for it, otherwise I would've forked it my self

Pointing me to a direction or making a branch with this enabled would be awesome

gnuton, 2022-10-18:
Hi,
Merlin comes already with (Generic routing encapsulation) GRE module available, so you do not need to reuild the kernel.
What they told you is to set CONFIG_NET_IPGRE=m in the kernel config and rebuild it in order to create the module you see here below.
Run the above commands on DSL-AC68U in order to be sure you have it.. I tested on DSL-AX82U.

gnuton@DSL-AX82U-FE18:/tmp/home/root# modprobe ip_gre
gnuton@DSL-AX82U-FE18:/tmp/home/root# lsmod | grep ip_gre
ip_gre                  8592  0

MHKalantarian, 2022-10-18:
Hey I think the problem was these modules are not available by default build. Also they closed GRE tunnels too :)) So can you guide me for `ipip` tunnel? I'd really appreciate it

Also here is the result of the commands showing I dont have the module by default:
```
admin@DSL-AC68U:/tmp/home/root# modprobe ip_gre
modprobe: module ip_gre not found in modules.dep
admin@DSL-AC68U:/tmp/home/root# lsmod | grep ip_gre
no response
```

MHKalantarian, 2022-10-18:
Here are my active modules if that helps
admin@DSL-AC68U:/tmp/home/root# lsmod
```
Module                  Size  Used by    Tainted: P
sr_mod                 10842  0
cdrom                  30901  1 sr_mod
thfsplus               84702  0
tntfs                 493645  0
tfat                  186782  0
ext2                   52776  0
ext4                  221674  1
crc16                   1007  1 ext4
jbd2                   48993  1 ext4
ext3                  106409  0
jbd                    42367  1 ext3
mbcache                 4599  3 ext2,ext4,ext3
usb_storage            34290  1
sg                     19855  0
sd_mod                 21983  2
scsi_wait_scan           416  0
scsi_mod              108738  4 sr_mod,usb_storage,sg,sd_mod
cdc_mbim                3137  0
qmi_wwan                5792  0
cdc_wdm                 7252  2 cdc_mbim,qmi_wwan
cdc_ncm                 8762  1 cdc_mbim
rndis_host              5256  0
cdc_ether               4036  1 rndis_host
asix                   10856  0
cdc_acm                12614  0
usbnet                 11900  6 cdc_mbim,qmi_wwan,cdc_ncm,rndis_host,cdc_ether,asix
mii                     3367  2 asix,usbnet
ohci_hcd               17926  0
ehci_hcd               32382  0
usbcore               102572 13 usb_storage,cdc_mbim,qmi_wwan,cdc_wdm,cdc_ncm,rndis_host,cdc_ether,asix,cdc_acm,usbnet,ohci_hcd,ehci_hcd
tdts_udbfw             26272  1
tdts_udb              181621  1 tdts_udbfw
tdts                  417542  2 tdts_udbfw,tdts_udb
nf_nat_sip              5031  0
nf_conntrack_sip       15713  1 nf_nat_sip
nf_nat_h323             4761  0
nf_conntrack_h323      33807  1 nf_nat_h323
nf_nat_rtsp             3202  0
nf_conntrack_rtsp       4067  1 nf_nat_rtsp
nf_nat_ftp              1144  0
nf_conntrack_ftp        4909  1 nf_nat_ftp
ip6table_mangle          934  0
ip6table_filter          750  1
nf_nat_pptp             1602  0
nf_conntrack_pptp       3419  1 nf_nat_pptp
nf_nat_proto_gre         887  1 nf_nat_pptp
nf_conntrack_proto_gre     3296  1 nf_conntrack_pptp
wl                   4202205  0
dpsta                  12782  1 wl
et                     65170  0
igs                    12040  1 wl
emf                    15449  2 wl,igs
ctf                    21199  0
jffs2                  91274  1
zlib_deflate           19489  1 jffs2
```