gnuton/asuswrt-merlin.ng #260: AiMesh BIND not working

toddgonzo74, 2022-10-15:
**Router Model Affected**
Models: AT8

**Firmware Version Affected**
**386.05_0-gnuton0_beta1**

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
In the AiMesh screen, I want to bind a client to a specific node.  In the stock firmware, you click bind and the bind button stays lit.  In this beta, when you try to bind, it says successful, prints the message, and then the binding isn't in effect once you click OK.  Binding doesn't appear to work. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to AiMesh
2. Click on Any client bind button (chain b utton to the right of the client MAC and IP).
3. Click bind and then click OK.  Wait a few seconds and you get a success message.
4. Click OK
5. note that client is not bound.

gnuton, 2022-10-15:
HI,
have you tried to reset to factory and try again? It should work (not tested no XT8, but on DSL-AX82U and it works)

gnuton, 2022-10-15:
> Not yet.. I will once all the bogons go to bed and I can thrash the
> network at will.
>
> Reconfig from scratch or config file?

reconfig from scratch. please do not use config files.

toddgonzo74, 2022-10-16:
That appears to have fixed it.  Gratzie.

gnuton, 2022-10-17:
I knew that! :D Awsesome!

closed 2022-10-17 by gnuton