gnuton/asuswrt-merlin.ng #262: New update alert on my ax5400 flashed to ax82u.  Do it from there or file?

dfgdgdgfd, 2022-10-17:
Is this normal for non betas?  I don't know if it's ofw or not.

gnuton, 2022-10-17:
not sure I got the problem.
I guess you have Aimesh with Ax5400 and Ax82.
I think if you have mixed environemnts OFW and CFW the CFW will update to CFW and the OFW will update to OFW but I am not sure about how the notifications would work.

dfgdgdgfd, 2022-10-17:
I don't have aimesh devices but I guess the router is capable of using them.

gnuton, 2022-10-17:
I apologie I though you were talking about the TUF-AX5400.
My mistake. You got an update alert because tehre is a new DSL-AX82U version available.

dfgdgdgfd, 2022-10-17:
Yes, I'm just not sure whether I can directly update from within the router or whether I have to download the update file from here and point the router to it.

gnuton, 2022-10-18:
If you have flashed my firmware the update notification is for a new version of my firmware. 
Merlin fimware must be downloaded and manually updated. So you cannot go wrong.