gnuton/asuswrt-merlin.ng #265: Stability adjustment upstream

trebleta, 2022-10-25:
Is it possable to add DSL stability adjustment to the upstream please, we have it for the downstream if it is possible.

gnuton, 2022-10-26:
not sure. it must be checked.
Is this available in stock firmware?

BoostOver, 2022-10-26:
@gnuton non è possibile la regolazione per nessun tipo di modem, l’unica possibile è quella inerente il download.
L’unico modo per stabilizzare l’upload è quella di chiamare l’operatore telefonico e farsi impostare un profilo diverso e in questo caso più performante con un snr basso…..perché mi sembra di intuire che l’utente voglia più banda!

gnuton, 2022-10-26:
not possible.

closed 2022-10-26 by gnuton

trebleta, 2022-10-26:
I see, as I did something and was able to drop the snr of the upstream by 2 gaining increased upload speed, just has to be able too ofc, UK ISPs are not helpfull

BoostOver, 2022-10-26:
@trebleta fluttuazioni fisiche della linea, semplicemente.

BoostOver, 2022-10-26:
@trebleta con il modem asus ac68u  è presente l’opzione UPBO cheti permettere di avere più banda in upload ma che non ha niente a che fare con l’snr.
Questo ti permette di avere più banda

trebleta, 2022-10-26:
Hmm I did not say I'm on a DSL-AX82U. And yes, was more due to a power cut in my area, droped it to 1db and increase bandwidth untill I rebooted the device