gnuton/asuswrt-merlin.ng #267: Nat Type incorrect?

trebleta, 2022-10-26:
Lastest @gnuton Fw fow DSL-AX82U-AX5400.
After flashing the new firmware with a hard and soft reset I tested the NAT type and it's full cone NAT, yet on offical firmware its symmetric Nat.

More info here
https://www.snbforums.com/threads/asuswrt-merlin-386-7-is-now-available-for-all-gnutons-models-too.80212/page-4
On page 3 at bottom were I went to test.

gnuton, 2022-10-27:
Was it correct in the previous release?

trebleta, 2022-10-27:
Could not say, as I've had to use the offical firmware for 2-3 months as I had a dsl and wifi problems.

trebleta, 2022-10-31:
Noticed its incorrect with the lastest offical firmware. So think it's something Asus have done

trebleta, 2022-10-31:
Yes, I did a hard factory reset.

gnuton, 2022-11-03:
did you try stock 3.0.0.4.386.49608 firmware right?
ASUS is now preparing the 388 images for my models.. let's keep this issue open and let's see if this issue is fixed there.

trebleta, 2022-11-04:
Yes, I have report to asus to. Also i tryed changing nat types, yet both report as normal NAT. Only think if I get time is try the last but 1 offical firmware. As the last had issues. See if I get what I use too.

trebleta, 2022-11-18:
Reply from asus is the sites I use are faulty now.