gnuton/asuswrt-merlin.ng #268: Full Spectrum of WiFi frequencies

broekhuijsen, 2022-10-26:
Hi there, I don't know if you can do it, as it probably is not legal, but I would like to see a feature to override the limitations in frequencies used for WiFi. There are less frequencies available than allowed, but besides that I would like to use the full potential of my hardware.

Is it possible to add this feature or to provide some sneaky patch at unofficially adds this feature? :D

gnuton, 2022-10-27:
The wifi is a closed source component and i do not have so much freedom
there. Maybe some users in the forum may know some magic tricks to unlock
something, but ATM i do not know anything about it.

broekhuijsen, 2022-10-27:
Ciao, thanks for responding so quickly. I might have found a way, but I will have to test that this weekend when I have more time and patience.

[Link to possible solution](https://gist.github.com/francoism90/3dede7973354d067c41bff5e54203fe9?permalink_comment_id=4002088#gistcomment-4002088)

But then my RT AX 55's still won't do because they don't have merlin software and stock cannot be manipulated in that way. 

Is there any chance you will support the RT AX 55 in the future? I know it is a budget router but it it is a great supplement to my XT8 and  could still benefit of some of the features right?

Any way, have a great night and thanks for your great work.

gnuton, 2022-10-28:
yes there may be chance to support it, but currently no plans.
I would have already adopted it if there were no V2 coming... but most of the router I support are getting revamped, so it sounds like I do need to double the amount of work for all routers now.. I wanna see how much this takes of my time...