gnuton/asuswrt-merlin.ng #270: Error when uploading firmware 386.08_0-gnuton1 to RT-AX92U

vdbseb1983, 2022-10-29:
**Router Model Affected**
RT-AX92

**Firmware Version Affected**
386.08_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
Error message when uploading firmware
Then goes back to recovery/safe image

**To Reproduce**
Steps to reproduce the behavior:
Click on manual upgrade file =>> Loading

**Expected behavior**
Expect to get file correct to update 

**Screenshots**
[image]

Other info:
[image]

Dump:
[Dump RT AX 92U.pdf](https://github.com/gnuton/asuswrt-merlin.ng/files/9894322/Dump.RT.AX.92U.pdf)

vdbseb1983, 2022-10-29:
Half solved: made error continue, as soon as returned directly to GUI: direct upload of image ok (du to less than 32mb), seems to check size of the "w" (ubi) image !

gnuton, 2022-12-27:
I will close this bug since it works. please re-openit if it still happens. I cannot reproduce this

closed 2022-12-27 by gnuton