gnuton/asuswrt-merlin.ng #272: RT-AX58U ASUSWrt Merlin Firmware Hakkında

Biyan21, 2022-11-02:
Merhaba,

RT-AX58U Ruter cihazıma ASUSWrt Merlin Firmware yüklemek istiyorum. Ruter arayüzünde Cake menüsüne ulaşmak istiyorum Ancak bir türlü yapamıyorum. Bana www uzantılı dosyayı gönderebilirmisiniz. Bende ruter cihazıma yükleme yapayım. Teşekürler.

gnuton, 2022-11-14:
I do not get a single word of what's written here. English please

Biyan21, 2022-11-15:
First of all, I apologize for not knowing English, so I wrote in Turkish.  I have a DSL-AX82U model device.  Modem, I want to install merlin on it.  So as with the RT-AX82U.

Biyan21, 2022-11-15:
I want to install ASUSWrt Merlin Firmware on my DSL-AX58U Modem Ruter.  I want to reach the Cake view in the modem interface, but I can't do it.  I wonder if there is no Merlin firmware for the modem?  Thanks.

gnuton, 2022-11-15:
Cake is only available on RT/DSL-AX devices and not on RT/DSL-AC ones, so no cake for DSL-AC68U.

closed 2022-11-15 by gnuton

Biyan21, 2022-11-15:
Thanks for the answer, I have the DSL-AX82U.

şunu yazdı:

gnuton, 2022-11-15:
DSL-AX82U supports cake and you can find hte firmware in the download page