gnuton/asuswrt-merlin.ng #273: Upload / Download speedometer into QoS page are inverted

WillyTP, 2022-11-03:
**Router Model Affected**
Models: (DSL-AX82U, maybe other)

**Firmware Version Affected**
386.08

**Is this bug present in upstream Merlin releases too?**
Don't know

**Describe the bug**
Into QoS page, description of /speedometer are inverted. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Adaptive QoS
2. Let something download...
3. Look at Speedometer named "upload". It actually should be download one.

**Expected behavior**
Download speedometer should refer to downloads, Upload speedometer to uploads

**Screenshots**
[image]

WillyTP, 2022-11-26:
Hi,
Looks like Traffic Analyzer is the cause.
By disabling it, upload and download speeds are NOT swapped anymore.

WillyTP, 2022-11-27:
First post updated.
BTW, I tried too to do another test. Going to Traffic Analyzer -> Statistics page caused Web UI to crash (GUI interface not available for some minutes...)
Rebooting the router solved WebUI issue.
I confirm again Traffic Analyzer <-> swapped speeds, I've been able to replicate them.

gnuton, 2022-11-28:
@WillyTP are you able to reproduce it? crashes btw is different issue.
If you are able to reproduce it all time please create a new ticket so that we can properly track it.

WillyTP, 2022-12-02:
Looks like issue has been solved in 388 Beta1.

closed 2022-12-02 by WillyTP