gnuton/asuswrt-merlin.ng #275: TUF-AX3000 applet astrillvpn

Chemyto, 2022-11-06:
**Router Model Affected**
TUF-AX3000

Firmware version
Stable: 386.07_2-gnuton1**Firmware Version Affected**

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
Dont install applet Astrillvpn

**To Reproduce**
Install Astrillvpn by SSH by does nothing

**Expected behavior**
Being able to install it as it happens in the RT-AX3000 model with firmware merlin

Thkanks

gnuton, 2022-11-08:
Hi,
I am not an astrill user. I could help you to get it working if you tell me what's the problem.
Could you please share more info and logs?

Chemyto, 2022-11-08:
Hi, thanks for you answer.

Wiki info install in asus merlin: https://www.astrill.com/wiki/Astrill_Setup_Manual:Getting_started_with_ASUS_Merlin_firmware_for_routers

Info install applet:
Astrill VPN works on certain DD-WRT, Tomato and ASUS Merlin routers. You have to install our applet over SSH. You may need to allow SSH in your router settings, so consult manual for your router OS before proceeding.

Once you get into SSH console on your router, run the following command:

eval `wget -q -O - http://astroutercn.com/router/install/3120955/xxxxxxxxxxxxxxxxxxxxxxxxx`
And that's it! If you now refresh your router admin page, you will see Astrill VPN page and can start using applet features.

You can find more information on how to set up the applet as well as router features on our wiki page