gnuton/asuswrt-merlin.ng #281: AiMesh nodes connected via ETH gets disconnected after a few days

WillyTP, 2022-11-10:
**Router Model Affected**
Models: (eg: DSL-AX82U/RT-AX82U/...)

**Firmware Version Affected**
386.01_0_gnuton1

**Is this bug present in upstream Merlin releases too?**
Don't know - it didn't happen with a AC86U I used before

**Describe the bug**
After a few days (no specific timeframe to report) AiMesh nodes get disconnected.
I have 2 of them, both connected via wire.

**To Reproduce**
Steps to reproduce the behavior:
Wait a few days, and see what happens...

**Expected behavior**
Well, this should not happen at all.

**Screenshots**
I'll post as soon as it happens again, I've already rebooted the router (way to temporarely fix it)

gnuton, 2022-11-10:
logs of when this happens could tell more.
when a nodes disconnects it should reconnect back. how long have you been waiting before reconnecting the router?
Are you sure the problem is not in the cable?

WillyTP, 2022-11-10:
Hi
This issue happened to me already two times.
Nodes gets disconnected (like as the cables had been disconnected), but obiouvsly nothing in physical has been done.
Cables are the usual one, a bit strange coincidence that it affects both two nodes.
I'll post more screens and logs should it happen again.

WillyTP, 2022-11-12:
Hi,
issue happened again today.

I've attached a screenshot plus the log.
Issue happened at about 10:30 (don't know the exact time), however in the log I see a kernel crash at that time.
[messages.log](https://github.com/gnuton/asuswrt-merlin.ng/files/9995378/messages.log)

Hope it helps.

[image]

WillyTP, 2022-11-16:
Hi again
I was using WPA2. I switched to WPA3 and waited for a few days (even tough, as I've said, nodes are connected via ETH).
Issue happened again.

I attach again the log.
Is there some tests I may do in order to debug this issue?
Is anyone else experiencing this?

[messages (1).log](https://github.com/gnuton/asuswrt-merlin.ng/files/10024268/messages.1.log)

@gnuton , should you like/need to connect to my router to further inquire, just tell me.

WillyTP, 2022-11-16:
I add too a log of router freshly rebooted, with everything working.
[messages (3).log](https://github.com/gnuton/asuswrt-merlin.ng/files/10024418/messages.3.log)

contrasenda, 2022-11-17:
This is also happening to me. Usually the node reconnects on its own quite quickly, but my kids yell nevertheless,.. 

My router is a RT-AX92U. Nodes are both RT-AC68U with official firmware. One of them is connected over ethernet, the other over wifi. Only the node over ethernet gets disconnected, and it never happened before flashing merlin.

I'll come back with logs next time it gets disconnected again.

WillyTP, 2022-11-27:
Hi
After upgrading to latest 388.1_0-gnuton0_alpha1, looks like this issue is not happening anymore.
However, is it worth to monitor this since root cause hasn't been identified.

WillyTP, 2022-11-28:
I close the ticket as requested.
Since 388 alpha / beta release, issue is not happening anymore.
Kind regards

WillyTP, 2022-12-02:
Hi,
after many days of proper working, today nodes have got disconnected again.
I haven't had time to save logs since I was in a hurry.
I'll provide more info when it happens again.
[image]

WillyTP, 2022-12-04:
Hi, happened again.
I attach logs.
[messages.log](https://github.com/gnuton/asuswrt-merlin.ng/files/10149686/messages.log)