gnuton/asuswrt-merlin.ng #284: Wrong behavior, trouble with hibernation on windows devices

Partizanin, 2022-11-18:
**Router Model Affected**
TUF AX3000

**Firmware Version Affected**
386.07_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
No information

**Describe the bug**
For some reason my laptop with windows doesn't go sleep or hibernate on this router but if i connect to another router "tplink" it go sleep normally as should. 

Probably it can be that router for some reason randomly send magic package to connected devices. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Connected devices to router go sleep normally as they should

gnuton, 2022-11-21:
I would rather check this issue in your windows driver.
AFAI ther are no magic packet keeping your network awake, Asus devices do have some wirless power management but that's actually should be the opposite of what happens to you. you can find the settings in the wireless page

Partizanin, 2022-11-21:
Thanks for your suggestion but i have already check everything, update driver and other, and as i told this problem appeared only when i connect to my tufax 3000 with 386.07_2-gnuton1 firmware, on tplink router it dosent happened.
So waiting for firmware update and hope that it will gone.