gnuton/asuswrt-merlin.ng #285: 388.1_0-gnuton0_alpha1 Feedback (TUF-AX5400)

swiffer, 2022-11-21:
Thanks releasing first alpha based on 388 series! I see several menus missing in the VPN section.

What is the difference between the two files / which one should be used?

- TUF-AX5400_388.1_0-gnuton0_alpha1_cferom_puresqubi.w
- TUF-AX5400_388.1_0-gnuton0_alpha1_puresqubi.w

**Router Model Affected**
TUF-AX5400

**Firmware Version Affected**
388.1_0-gnuton0_alpha1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
VPN: menus (e.g. when on /Advanced_VPN_OpenVPN.asp) and/or menu entries (VPN Director) are missing

**Screenshots**
[image]
[image]
[image]
[image]
[image]

gnuton, 2022-11-21:
Thanks for reporting,
TUF-AX5400_388.1_0-gnuton0_alpha1_cferom_puresqubi.w <--- with  CFE (pronuced cafe', it's the Broadcom bootloader)
TUF-AX5400_388.1_0-gnuton0_alpha1_puresqubi.w <--  this doesn't update the bootloader

I would most probably get rid of TUF ui in 388 and I am not sure if I would make TUF UI enabled images too.
This first alphas do still has TUF UI.
I will try to collect feedback for it and take a decision...

swiffer, 2022-11-21:
For me would be fine to use the default ui. Rmerl seems to offer rog Variants in addition now.

Ok, thanks - is one or the other recommended? Do you know what happens when using the default Firmware? (Incl. Or excl. The cferom)?

DervilRus, 2022-11-21:
I also do not know which option is better in which case

micron888, 2022-11-28:
TUF-AX5400_388.1_0-gnuton0_beta1_tuf_puresqubi.w - dnsfilter and VPN director broken 
TUF-AX5400_388.1_0-gnuton0_beta1_puresqubi.w - dnsfilter replace with dns director. VPN tab is available.

gnuton, 2022-11-30:
@micron888 so TUF is broken and the standard gui looks fine, correct?

micron888, 2022-11-30:
Standard gui look ok, other than some cosmetic issue indicated on #292

Rovasteen, 2022-12-01:
For CFE vs non-CFE, there is always risk when you flash the CFE bootloader.  It would be nice to know if the CFE is actually being updated in those builds because flashing the same CFE again adds risk with no benefit.  If the CFE Is actually being updated, it would be nice to get some notes which state what is in the CFE update.

gnuton, 2023-09-07:
close as it's an old issue not present in the latest releases

closed 2023-09-07 by gnuton