gnuton/asuswrt-merlin.ng #286: tuf-ax3000无线网络标签有问题

qqw1231, 2022-11-21:
无线网络，标签有问题
[image]

qqw1231, 2022-11-21:
中文标签有问题

gnuton, 2022-11-23:
Hi,
thanks for reporting.
If I got it right you are complaining about the chinese localization of the WiFi SSID and WPA-PSK.
Those localization come from asus.
Can you be more explicit with the problem? I am not chinese speaking.
Do you see the same problem in upstream firmware?

qqw1231, 2022-11-23:
> Hi, thanks for reporting. If I got it right you are complaining about the chinese localization of the WiFi SSID and WPA-PSK. Those localization come from asus. Can you be more explicit with the problem? I am not chinese speaking. Do you see the same problem in upstream firmware?

There is no such problem in the official firmware

I see your updated content is about to replace the TUF UI. I also think the official TUF is not good. I hope you can replace the TUF UI

qqw1231, 2022-11-23:
> Hi, thanks for reporting. If I got it right you are complaining about the chinese localization of the WiFi SSID and WPA-PSK. Those localization come from asus. Can you be more explicit with the problem? I am not chinese speaking. Do you see the same problem in upstream firmware?

Will the next firmware TUF UI be changed to this one?
[image]

qqw1231, 2022-11-23:
> Hi, thanks for reporting. If I got it right you are complaining about the chinese localization of the WiFi SSID and WPA-PSK. Those localization come from asus. Can you be more explicit with the problem? I am not chinese speaking. Do you see the same problem in upstream firmware?

Replace the ROG UI with the next firmware of the TUF AX3000?

gnuton, 2022-11-24:
there may be chance I will get rid of TUF_UI yes.
I am not sure yet what's the problem here btw :D

qqw1231, 2022-11-24:
> there may be chance I will get rid of TUF_UI yes. I am not sure yet what's the problem in this bug report btw. is it localization or what? If so what should it be?

There is no problem when you change to English language, and there will be problems when you change to Chinese. Is it because the last row of characters is too long? WiFi Radar. English only has WiFi Radar. Wait until you change the TUF UI to the ROG UI

gnuton, 2022-11-24:
in chrome if you right click the text and you press inspect you can see if it the chinese word gets truncated.
In that case it's a CSS issue, otherwise is a localzation issue.