gnuton/asuswrt-merlin.ng #288: The WPS tab is missing.

hammerjoe, 2022-11-22:
**Router Model Affected** RT-AX82U
Models: (eg: DSL-AX82U/RT-AX82U/...)

**Firmware Version Affected**
[388.1_0-gnuton0_alpha1](https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/388.1_0-gnuton0_alpha1)

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
The WPS tab is missing from the WAN options.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section

gnuton, 2022-11-23:
Hi,
The WPS tab is in Wireless 
http://router.asus.com/Advanced_WWPS_Content.asp
and that's the right place for it.

gnuton, 2022-11-23:
[image]

closed 2022-11-23 by gnuton

hammerjoe, 2022-11-23:
Thanks. 
I was pretty sure I checked all the tabs but somehow missed it...

commit a163ba7e (2019-05-09): nano: update to 4.0 (closes #288)