gnuton/asuswrt-merlin.ng #289: XT8 AiMesh node disconnects and reconnects every 12 hours on the dot...

sodaboy581, 2022-11-22:
So, I have 2 XT8s in an AiMesh.

Both are running 386.07_2-gnuton1.

I've noticed that the AiMesh node will disconnect every 12 hours EXACTLY and reconnect to the AiMesh router.

Is this just something the firmware does?

Also, if so, is there a way to disable this disconnect and reconnect?

It's not a huge deal and everything is solid otherwise.  It's just an interesting quirk!

Power is NOT lost during this time and I don't have any other scheduled event going on!

Here's some clips from the log (Mac addresses edited):

Nov 19 09:26:10 wlceventd: wlceventd_proc_event(469): eth4: Deauth_ind 7C:XX:XX:XX:XX:01, status: 0, reason: Station requesting (re)association is not authenticated with responding station (9)
Nov 19 09:26:10 wlceventd: wlceventd_proc_event(486): eth4: Disassoc 7C:XX:XX:XX:XX:01, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
Nov 19 09:26:10 kernel: Flushing net_device wds0.0.1.
Nov 19 09:26:10 wlceventd: wlceventd_proc_event(534): eth6: Assoc 7C:XX:XX:XX:XX:08, status: Successful (0)
Nov 19 09:26:10 kernel: Register interface [wds2.0.1]  MAC: 7c:XX:XX:XX:XX:c8
Nov 19 09:26:11 wlceventd: wlceventd_proc_event(534): eth4: Assoc 7C:XX:XX:XX:XX:01, status: Successful (0)
Nov 19 21:26:11 wlceventd: wlceventd_proc_event(486): wds2.0.1: Disassoc 7C:XX:XX:XX:XX:08, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
Nov 19 21:26:11 wlceventd: wlceventd_proc_event(486): eth6: Disassoc 7C:XX:XX:XX:XX:08, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
Nov 19 21:26:29 wlceventd: wlceventd_proc_event(534): eth6: Assoc 7C:XX:XX:XX:XX:08, status: Successful (0)
Nov 19 21:26:29 kernel: Register interface [wds2.0.1]  MAC: 7c:XX:XX:XX:XX:c8
Nov 20 09:26:30 wlceventd: wlceventd_proc_event(486): wds2.0.1: Disassoc 7C:XX:XX:XX:XX:08, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
Nov 20 09:26:30 wlceventd: wlceventd_proc_event(486): eth6: Disassoc 7C:XX:XX:XX:XX:08, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
Nov 20 09:26:49 wlceventd: wlceventd_proc_event(534): eth4: Assoc 7C:XX:XX:XX:XX:01, status: Successful (0)
Nov 20 09:26:50 wlceventd: wlceventd_proc_event(534): eth6: Assoc 7C:XX:XX:XX:XX:08, status: Successful (0)
Nov 20 09:26:50 kernel: Register interface [wds2.0.1]  MAC: 7c:XX:XX:XX:XX:c8
Nov 20 21:26:50 wlceventd: wlceventd_proc_event(486): wds0.0.1: Disassoc 7C:XX:XX:XX:XX:01, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
Nov 20 21:26:50 wlceventd: wlceventd_proc_event(486): eth4: Disassoc 7C:XX:XX:XX:XX:01, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
Nov 20 21:26:50 kernel: Flushing net_device wds0.0.1.
Nov 20 21:26:51 wlceventd: wlceventd_proc_event(486): wds2.0.1: Disassoc 7C:XX:XX:XX:XX:08, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
Nov 20 21:26:51 wlceventd: wlceventd_proc_event(486): eth6: Disassoc 7C:XX:XX:XX:XX:08, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
Nov 20 21:27:10 wlceventd: wlceventd_proc_event(534): eth4: Assoc 7C:XX:XX:XX:XX:01, status: Successful (0)
Nov 20 21:27:11 wlceventd: wlceventd_proc_event(534): eth6: Assoc 7C:XX:XX:XX:XX:08, status: Successful (0)
Nov 20 21:27:11 kernel: Register interface [wds2.0.1]  MAC: 7c:XX:XX:XX:XX:c8
Nov 21 09:27:11 wlceventd: wlceventd_proc_event(486): wds0.0.1: Disassoc 7C:XX:XX:XX:XX:01, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
Nov 21 09:27:11 wlceventd: wlceventd_proc_event(486): eth4: Disassoc 7C:XX:XX:XX:XX:01, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
Nov 21 09:27:11 kernel: Flushing net_device wds0.0.1.
Nov 21 09:27:12 wlceventd: wlceventd_proc_event(486): wds2.0.1: Disassoc 7C:XX:XX:XX:XX:08, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
Nov 21 09:27:12 wlceventd: wlceventd_proc_event(486): eth6: Disassoc 7C:XX:XX:XX:XX:08, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
Nov 21 09:27:30 wlceventd: wlceventd_proc_event(534): eth4: Assoc 7C:XX:XX:XX:XX:01, status: Successful (0)
Nov 21 09:27:32 wlceventd: wlceventd_proc_event(534): eth6: Assoc 7C:XX:XX:XX:XX:08, status: Successful (0)
Nov 21 09:27:32 kernel: Register interface [wds2.0.1]  MAC: 7c:XX:XX:XX:XX:c8
Nov 21 21:27:31 wlceventd: wlceventd_proc_event(486): wds0.0.1: Disassoc 7C:XX:XX:XX:XX:01, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
Nov 21 21:27:31 wlceventd: wlceventd_proc_event(486): eth4: Disassoc 7C:XX:XX:XX:XX:01, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
Nov 21 21:27:31 kernel: Flushing net_device wds0.0.1.
Nov 21 21:27:33 wlceventd: wlceventd_proc_event(486): wds2.0.1: Disassoc 7C:XX:XX:XX:XX:08, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
Nov 21 21:27:33 wlceventd: wlceventd_proc_event(486): eth6: Disassoc 7C:XX:XX:XX:XX:08, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
Nov 21 21:27:51 wlceventd: wlceventd_proc_event(534): eth4: Assoc 7C:XX:XX:XX:XX:01, status: Successful (0)
Nov 21 21:27:53 wlceventd: wlceventd_proc_event(534): eth6: Assoc 7C:XX:XX:XX:XX:08, status: Successful (0)
Nov 21 21:27:53 kernel: Register interface [wds2.0.1]  MAC: 7c:XX:XX:XX:XX:c8
Nov 22 09:27:52 wlceventd: wlceventd_proc_event(486): wds0.0.1: Disassoc 7C:XX:XX:XX:XX:01, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
Nov 22 09:27:52 wlceventd: wlceventd_proc_event(486): eth4: Disassoc 7C:XX:XX:XX:XX:01, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
Nov 22 09:27:52 kernel: Flushing net_device wds0.0.1.
Nov 22 09:27:54 wlceventd: wlceventd_proc_event(486): wds2.0.1: Disassoc 7C:XX:XX:XX:XX:08, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
Nov 22 09:27:54 wlceventd: wlceventd_proc_event(486): eth6: Disassoc 7C:XX:XX:XX:XX:08, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
Nov 22 09:28:10 wlceventd: wlceventd_proc_event(534): eth4: Assoc 7C:XX:XX:XX:XX:01, status: Successful (0)
Nov 22 09:28:12 wlceventd: wlceventd_proc_event(534): eth6: Assoc 7C:XX:XX:XX:XX:08, status: Successful (0)
Nov 22 09:28:12 kernel: Register interface [wds2.0.1]  MAC: 7c:XX:XX:XX:XX:c8

gnuton, 2022-11-23:
Can you please highlight the lines that show this behaviour?

sodaboy581, 2022-11-23:
Those were the highlights.

I extracted those lines manually from my log.

The 7C:XX:XX addresses are the addresses of the AiMesh node's different radios.

gnuton, 2022-11-24:
wlceventd: wlceventd_proc_event(486): eth6: Disassoc 7C:XX:XX:XX:XX:08, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
this means the node has disconnected itself from the master.
Do you have the logs of the nodes? 
With my current understanding, usually planned disconnections happens for two reasons:
- energy saving 
- rebalancing
- or forced by asus in order to reset the connection in case semething goes wrong.

BTW these are debug messages... so not so important.
If you wanna test something you should continuosly ping the master node from a node
(eg you can ssh int the node and run screen - install it if you do not have it) and from ther run the continuos pings. or just connect your pc to the node and run the ping from your pc to the master node.
In theory you should not lose the connection anymore if this happens.

snacques, 2023-01-05:
I am seeing the same issue on a wireless remote XT8 station. Similar log messages AND all devices that were previously connected to that node are disconnected for 30-60 second while they reconnect to another node.

gnuton, 2023-01-05:
@snacques the lines you see above are debug messages. not error and not even warnings.
They are just informative messages that log the status of network.
If the device disconnect and then reconnect, there is a reason.
There may be multiple good reason this could happen by design.

snacques, 2023-01-05:
If you can point me to instructions on getting the remote node logs, I'll happily pull them on the next event.

rlfrank165, 2023-01-07:
I'm offering these in the hope that others more knowledgeable may find
something useful here.

I too have been plagued by repeated crashes and reboots, about one a day,
but not on any reguilar schedule.  From my limited perspective I believe
these have nothing to do with the Merlin releases.

My experience is that  when I begin using the 388 official releases about
two months ago from Asus in my 3-node, wired ethernet AIMesh XT8
configuration these problems began.  In fact, they became so bad I finally
used that as a reason to start using Merlin, and I went back to the 386
release of Gnuton1 Merlin which was all that was available for the XT8 at
that time, which radically improved my stability.  It was only when I
"upgraded" to the 388 based Gnuton1 Merlin releases (beta1 and then beta2)
that things became unstable again. So believe these issues are inheirited
from the 388 based Asus GPL release and not from anything that the official
Merlin and Gnuton1 Merlin additions to the 388 GPL from Asus have done.

Having said that, given there is no one inside of Asus who I can
communicate with, others in this community are better connected and much
more technically literate, we may be doing a service to the overall Asus
user base if we have any insight into what in the official Asus releases
may be behind these issues.  I should also point out that as I look at the
official Asus Community blogs, people there are also complaining left and
right about instability introduced in the various 388 based firmware
upgrades.

So with this in mind, I am attaching two of the most recent syslogs
surrounding crashes over the last few days (1/3 and 1/5) beginning with an
hour or so before the crash and only including a small section of the after
reboot syslog.  If more before or after context is needs, I still have the
full syslogs and I can include as much pre and post crash syslog detail as
is desired.  I have added the word CRASH and the point where I believe the
routers rebooted.

Once again, I wish I was more technically proficient in being able to look
at these syslogs.  I do keep syslogs for about a month, so if these are
useful I can go back and provide some older crash syslogs as well.  I do
plan on upgrading to beta3 later tonight.

I really appreciate the enormous efforts of Gnuton in helping to produce
this amazing resource.

Randy Frank

Daemon.InfoJan  3 06:07:47 Frank-A2Router-BB276DE-C hostapd: eth5: STA 98:48:27:78:bd:69 WPA: group key handshake completed (RSN)
Daemon.InfoJan  3 06:07:47 Frank-A2Router-BB276DE-C hostapd: eth5: STA 4c:ef:c0:75:cd:e4 WPA: group key handshake completed (RSN)
Daemon.InfoJan  3 06:07:47 Frank-A2Router-BB276DE-C dnsmasq-dhcp[31952]: DHCPREQUEST(br0) 192.168.10.138 f0:2f:74:59:c0:70 
Daemon.InfoJan  3 06:07:47 Frank-A2Router-BB276DE-C dnsmasq-dhcp[31952]: DHCPACK(br0) 192.168.10.138 f0:2f:74:59:c0:70 ZenWiFi_XT8-C070
Daemon.InfoJan  3 06:07:47 Frank-A2Router-BB276DE-C hostapd: eth5: STA 44:61:32:33:ec:93 WPA: group key handshake completed (RSN)
Daemon.InfoJan  3 06:07:47 Frank-A2Router-BB276DE-C hostapd: eth5: STA 08:12:a5:48:72:c5 WPA: group key handshake completed (RSN)
Daemon.InfoJan  3 06:07:47 Frank-A2Router-BB276DE-C hostapd: eth5: STA a0:02:dc:e9:07:45 WPA: group key handshake completed (RSN)
Daemon.InfoJan  3 06:07:48 Frank-A2Router-BB276DE-C hostapd: eth6: STA 84:5f:04:9b:b4:19 WPA: group key handshake completed (RSN)
Daemon.InfoJan  3 06:07:48 Frank-A2Router-BB276DE-C hostapd: eth4: STA 2c:9f:fb:8a:53:44 WPA: group key handshake completed (RSN)
Daemon.InfoJan  3 06:07:48 Frank-A2Router-BB276DE-C hostapd: eth4: STA 9c:8e:cd:0e:da:b3 WPA: group key handshake completed (RSN)
Daemon.InfoJan  3 06:07:48 Frank-A2Router-BB276DE-C hostapd: eth4: STA b8:b7:d7:06:eb:9f WPA: group key handshake completed (RSN)
Daemon.InfoJan  3 06:07:48 Frank-A2Router-BB276DE-C hostapd: eth4: STA b4:b0:24:ea:88:d7 WPA: group key handshake completed (RSN)
Daemon.InfoJan  3 06:07:48 Frank-A2Router-BB276DE-C hostapd: eth4: STA b4:b0:24:2a:3d:39 WPA: group key handshake completed (RSN)
Daemon.InfoJan  3 06:07:48 Frank-A2Router-BB276DE-C hostapd: eth4: STA c4:6e:7b:a3:65:ae WPA: group key handshake completed (RSN)
Daemon.InfoJan  3 06:07:48 Frank-A2Router-BB276DE-C hostapd: eth4: STA 2c:aa:8e:1e:63:08 WPA: group key handshake completed (RSN)
Daemon.InfoJan  3 06:07:48 Frank-A2Router-BB276DE-C hostapd: eth4: STA 60:a4:b7:9f:4e:06 WPA: group key handshake completed (RSN)
Daemon.InfoJan  3 06:07:48 Frank-A2Router-BB276DE-C hostapd: eth4: STA 60:a4:b7:82:59:97 WPA: group key handshake completed (RSN)
Daemon.InfoJan  3 06:07:48 Frank-A2Router-BB276DE-C hostapd: eth4: STA 00:5f:67:af:00:c0 WPA: group key handshake completed (RSN)
Daemon.InfoJan  3 06:07:48 Frank-A2Router-BB276DE-C hostapd: eth4: STA 00:5f:67:af:0f:c8 WPA: group key handshake completed (RSN)
Daemon.InfoJan  3 06:07:48 Frank-A2Router-BB276DE-C hostapd: eth4: STA 00:31:92:d9:bd:f8 WPA: group key handshake completed (RSN)
Daemon.InfoJan  3 06:07:48 Frank-A2Router-BB276DE-C hostapd: eth4: STA 00:5f:67:ae:ff:0e WPA: group key handshake completed (RSN)
Daemon.InfoJan  3 06:07:48 Frank-A2Router-BB276DE-C hostapd: eth4: STA b4:b0:24:4c:3c:39 WPA: group key handshake completed (RSN)
Daemon.InfoJan  3 06:07:48 Frank-A2Router-BB276DE-C hostapd: eth4: STA c0:06:c3:91:07:02 WPA: group key handshake completed (RSN)
Daemon.InfoJan  3 06:07:48 Frank-A2Router-BB276DE-C hostapd: eth4: STA b4:b0:24:33:9f:7f WPA: group key handshake completed (RSN)
[... cut ...]
User.ErrorMay  5 01:05:31 Frank-A2Router-BB276DE-C acsd: acsd_main_loop(1176): sync_id: 25337, status:1, misc.error/abort
User.NoticeMay  5 01:05:31 Frank-A2Router-BB276DE-C wlceventd: wlceventd_proc_event(534): eth4: Assoc D8:F1:5B:B4:6D:0B, status: Successful (0)
User.NoticeMay  5 01:05:31 Frank-A2Router-BB276DE-C wlceventd: wlceventd_proc_event(505): eth4: Auth C4:4F:33:B7:0D:69, status: Successful (0)
Daemon.InfoMay  5 01:05:31 Frank-A2Router-BB276DE-C hostapd: eth4: STA d8:f1:5b:b4:6d:0b IEEE 802.11: associated

snacques, 2023-01-07:
I had the same experience with 388, which is why I'm here. But this thread is about 386.07_2-gnuton1, the version I and the OP are on.