gnuton/asuswrt-merlin.ng #290: DSL-AX82U DSL modem doesn't work if SNR values are modified

gnuton, 2022-11-24:
**Router Model Affected**
DSL-AX82U

**Firmware Version Affected**
388.01 alpha1

**Is this bug present in upstream Merlin releases too?**
no
**Describe the bug**
Some users noticed issues with the DSL modem if SNR is changed.
Factory reset helps to ripristinate the modem.

gnuton, 2022-11-26:
DSL Modems seems not to correctly work when SNR is set to something different than 0

-dslx_snrm_offset=0
+dslx_snrm_offset=112

In order for the modem to get seen again, you want to set back the SNR to disabled and then reboot the unit, this makes all working again.

Root cause of this is under investigation.

gnuton, 2022-11-26:
Inolved kernel modules
/proc/sys/net/ipv4/conf/dsl0# lsmod | grep dsl
adsldd                512980  0 
bcmxtmcfg              95175  1 adsldd

gnuton, 2022-11-26:
rgrep dslx_snrm_offset
Binary file rc/prebuild/DSL-AX82U/dsl_fb.o matches
Binary file rc/prebuild/DSL-AC68U/dsl_fb.o matches
rc/dsld.c:	cur_record->snrm = nvram_get_int("dslx_snrm_offset");
rc/dsl.c:	p->snrm = nvram_get_int("dslx_snrm_offset");
shared/defaults.c:	{ "dslx_snrm_offset", "0", CKN_STR8, CKN_TYPE_DEFAULT, CKN_ACC_LEVEL_DEFAULT, CKN_ENC_DEFAULT, 0 },	/* Paul add 2012/9/24 */
Binary file dsl_drv_tool/tp_init/prebuild/tp_init matches

gnuton, 2022-11-28:
I have noticed all images comes with two DSL drivers.
one is the newest one and the other one ins /etc/adsl1 is an old one which is used as fallback in case SNR is changed.
this driver is A2pv6L047f1, 
The reason for having two drivers is that chipset vendors will not support this feature in new driver since it is non-statndard.

gnuton, 2022-11-29:
root@DSL-AX82U:/tmp/home/root# md5sum /rom/etc/adsl/adsl_phy.bin
b571853775ffaf3751627ca3fab3bb7d  /rom/etc/adsl/adsl_phy.bin
root@DSL-AX82U:/tmp/home/root# md5sum /rom/etc/adsl1/adsl_phy.bin
46060171e3674dfca17e72342f8fc877  /rom/etc/adsl1/adsl_phy.bin

gnuton, 2022-12-09:
not working driver
# find / -name "adsl*"
/lib/modules/4.1.52/extra/adsldd.ko
/rom/etc/adsl
/rom/etc/adsl/adsl_phy.bin
/rom/etc/adsl1
/rom/etc/adsl1/adsl_phy.bin
/tmp/adslcfg.dat
/tmp/adsl
/tmp/etc/adsl

gnuton, 2022-12-09:
working driver
```
 find / -name adsl*
/lib/modules/4.1.52/extra/adsldd.ko
/rom/etc/adsl
/rom/etc/adsl/adsl_phy.bin
/rom/etc/adsl1
/rom/etc/adsl1/adsl_phy.bin
/sys/module/adsldd
/sys/module/bcmxtmcfg/holders/adsldd
/tmp/adsl
/tmp/etc/adsl1
/tmp/etc/adsl

```

gnuton, 2022-12-09:
diff ./release/src-rt-5.04axhnd.675x/bcmdrivers/opensource/include/bcm963xx/bcmxtmcfg.h /home/aaloisio/GITonio/ASUSWRT_ORIG/__ORIG/NEW/asuswrt/release/src-rt-5.02axhnd.675x/bcmdrivers/opensource/include/bcm963xx/bcmxtmcfg.h

gnuton, 2022-12-17:
/sys/devices/virtual/net/dsl0# ls
addr_assign_type   dev_port           link_mode          queues
addr_len           dormant            mtu                speed
address            duplex             name_assign_type   statistics
bcm_mcastrouter    flags              netdev_group       subsystem
broadcast          gro_flush_timeout  operstate          tx_queue_len
carrier            ifalias            phys_port_id       type
carrier_changes    ifindex            phys_port_name     uevent
dev_id             iflink             phys_switch_id

gnuton, 2022-12-17:
# find / -name dsl0 -print
/proc/sys/net/ipv4/conf/dsl0
/proc/sys/net/ipv4/neigh/dsl0
find: /proc/7900: No such file or directory
/sys/devices/virtual/net/dsl0
/sys/class/net/dsl0

gnuton, 2023-01-04:
I think I found the issue.
The problem was in the a tool which builds the ROM filesystem and that for some reason was not correctly with the old DSL firmware.

This is the latest beta image with the SNR fix.
To test it just enable "Stability adjustement" and check if you router can connect to internet through DSL
[image]

https://output.circle-artifacts.com/output/job/4309befa-b3aa-43d0-af6f-a86183a1405b/artifacts/0/workspace/release/DSL-AX82U_388.1_0_puresqubi.w

JustifYI, 2023-01-05:
> I think I found the issue. The problem was in the a tool which builds the ROM filesystem and that for some reason was not correctly with the old DSL firmware.
>
> This is the latest beta image with the [...]

Great, changing the SNR now works properly!

djt7787, 2023-01-06:
Yeah! Works perfectly on mine too, seems to stable. Thanks again gnuton 🍻

gnuton, 2023-01-06:
sorry for the wait guys this was bug was A LOT of digging....

gnuton, 2023-01-06:
the fix is now in the master line and it will be part of the next images

commit 9cd08eee (2019-05-09): rc: provide filesystem type as second argument to pre-mount script (closes #290)
closed 2023-01-06 by gnuton