gnuton/asuswrt-merlin.ng #291: tuf ax3000  Unstable: 388.1_0-gnuton0_beta1

qqw1231, 2022-11-28:
The background of the router can not be opened and the network is not available. It will be fine after a long time. Many times

qqw1231, 2022-11-28:
When watching the live broadcast, the network is disconnected. The background of the router cannot be opened. Sometimes you need to turn off the mobile phone wifi to watch the live broadcast from a new connection

qqw1231, 2022-11-28:
It seems that the live broadcast has been disconnected continuously. The background of the router cannot be opened. It appears many times. Sometimes you need to turn off the mobile phone wifi to watch the live broadcast again from a new connection

gnuton, 2022-11-29:
Hi,
1 what's the background of the router? If you are talking about the WEB interface, please open it incognito expecially if you have  flashed an image without TUF ui.
2 What's the live broadcast ? can you elaborate more?

qqw1231, 2022-11-29:
Live Network, with the latest firmware will be off the network, continuous appearance, watching live when off the network live will be inexplicably suspended