gnuton/asuswrt-merlin.ng #294: TUF AX3000 Unstable: 388.1_0-gnuton0_beta1

qqw1231, 2022-12-01:
The network is suddenly disconnected after a while, which is very frequent. I hope it can be repaired

gnuton, 2022-12-01:
please factory reset the router and check if this happens again.

qqw1231, 2022-12-01:
> please factory reset the router and check if this happens again.

This problem occurs after the factory settings

gnuton, 2022-12-01:
ok how can you plese write down how I can reproduce it?

qqw1231, 2022-12-01:
> ok how can you plese write down how I can reproduce it? do you also have logs?

May  5 05:05:15 syslogd started: BusyBox v1.25.1
May  5 05:05:15 kernel: klogd started: BusyBox v1.25.1 (2022-11-27 23:11:22 UTC)
May  5 05:05:15 kernel: Booting Linux on physical CPU 0x0
May  5 05:05:15 kernel: Linux version 4.1.52 (docker@8f37e2323af8) (gcc version 5.5.0 (Buildroot 2017.11.1) ) #1 SMP PREEMPT Sun Nov 27 23:16:25 UTC 2022
May  5 05:05:15 kernel: CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
May  5 05:05:15 kernel: CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
May  5 05:05:15 kernel: Machine model: Broadcom BCM963178
May  5 05:05:15 kernel: bootconsole [earlycon0] enabled
May  5 05:05:15 kernel: Memory policy: Data cache writealloc
May  5 05:05:15 kernel: PERCPU: Embedded 10 pages/cpu @dfbd2000 s11852 r8192 d20916 u40960
May  5 05:05:15 kernel: Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 130048
May  5 05:05:15 kernel: Kernel command line: root=/dev/ubiblock0_0 ubi.mtd=0 ubi.block=0,0 rootfstype=squashfs console=ttyAMA0 earlyprintk debug irqaffinity=0 pci=pcie_bus_safe
May  5 05:05:15 kernel: PID hash table entries: 2048 (order: 1, 8192 bytes)
May  5 05:05:15 kernel: Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
May  5 05:05:15 kernel: Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
May  5 05:05:15 kernel: Memory: 511988K/524288K available (4511K kernel code, 198K rwdata, 1220K rodata, 208K init, 410K bss, 12300K reserved, 0K cma-reserved, 0K highmem)
May  5 05:05:15 kernel: Virtual kernel memory layout:
May  5 05:05:15 kernel:     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
May  5 05:05:15 kernel:     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
May  5 05:05:15 kernel:     vmalloc : 0xe0800000 - 0xff000000   ( 488 MB)
May  5 05:05:15 kernel:     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
May  5 05:05:15 kernel:     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
May  5 05:05:15 kernel:     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
May  5 05:05:15 kernel:       .text : 0xc0018000 - 0xc05b11b0   (5733 kB)
May  5 05:05:15 kernel:       .init : 0xc05b2000 - 0xc05e6000   ( 208 kB)
May  5 05:05:15 kernel:       .data : 0xc05e6000 - 0xc06179a0   ( 199 kB)
May  5 05:05:15 kernel:        .bss : 0xc06179a0 - 0xc067e384   ( 411 kB)
May  5 05:05:15 kernel: Preemptible hierarchical RCU implementation.
May  5 05:05:15 kernel: NR_IRQS:16 nr_irqs:16 16
May  5 05:05:15 kernel: Architected cp15 timer(s) running at 50.00MHz (phys).
May  5 05:05:15 kernel: clocksource arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb8812736b, max_idle_ns: 440795202655 ns
May  5 05:05:15 kernel: sched_clock: 56 bits at 50MHz, resolution 20ns, wraps every 4398046511100ns
May  5 05:05:15 kernel: Switching to timer-based delay loop, resolution 20ns
May  5 05:05:15 kernel: BRCM Legacy Drivers' Helper, all legacy drivers' IO memories/interrupts should be remapped here
May  5 05:05:15 kernel:      Remapping interrupts...
May  5 05:05:15 kernel:              hwirq      virq
May  5 05:05:15 kernel:                60        20
May  5 05:05:15 kernel:                61        21
May  5 05:05:15 kernel:                43        22
May  5 05:05:15 kernel:                44        23
May  5 05:05:15 kernel:                45        24
May  5 05:05:15 kernel:                46        25
May  5 05:05:15 kernel:                66        26
May  5 05:05:15 kernel:               110        27
May  5 05:05:15 kernel:                69        28
May  5 05:05:15 kernel:               152        29
May  5 05:05:15 kernel:               153        30
May  5 05:05:15 kernel:               154        31
May  5 05:05:15 kernel:               155        32
May  5 05:05:15 kernel:               156        33
May  5 05:05:15 kernel:               157        34
May  5 05:05:15 kernel:               158        35
May  5 05:05:15 kernel:               159        36
May  5 05:05:15 kernel:                76        37
May  5 05:05:15 kernel:                78        38
May  5 05:05:15 kernel:                77        39
May  5 05:05:15 kernel:                79        40
May  5 05:05:15 kernel:               104        41
May  5 05:05:15 kernel:               105        42
May  5 05:05:15 kernel:               106        43
May  5 05:05:15 kernel:                96        44
May  5 05:05:15 kernel:                97        45
May  5 05:05:15 kernel:                98        46
May  5 05:05:15 kernel:                99        47
May  5 05:05:15 kernel:               100        48
May  5 05:05:15 kernel:               101        49
May  5 05:05:15 kernel:               148        50
May  5 05:05:15 kernel:                71        51
[... 1364 lines cut ...]
Dec  1 18:44:38 acsd: acs_candidate_score_intf(1080): eth6: intf check failed for chanspec: 0xe832
Dec  1 18:44:38 acsd: acs_candidate_score_bgnoise(1500): eth6: bgnoise check failed for chanspec: 0xe832
Dec  1 18:44:38 acsd: acs_candidate_score_txop(1758): eth6: txop check failed for chanspec: 0xe832
Dec  1 18:44:38 acsd: acs_candidate_score_intf(1080): eth6: intf check failed for chanspec: 0xe932
Dec  1 18:44:38 acsd: acs_candidate_score_bgnoise(1500): eth6: bgnoise check failed for chanspec: 0xe932
Dec  1 18:44:38 acsd: acs_candidate_score_txop(1758): eth6: txop check failed for chanspec: 0xe932
Dec  1 18:44:38 acsd: acs_candidate_score_intf(1080): eth6: intf check failed for chanspec: 0xea32
[... cut ...]
Dec  1 19:02:02 acsd: acs_candidate_score_bgnoise(1500): eth6: bgnoise check failed for chanspec: 0xea32
Dec  1 19:02:02 acsd: acs_candidate_score_txop(1758): eth6: txop check failed for chanspec: 0xea32
Dec  1 19:02:02 acsd: acs_candidate_score_intf(1080): eth6: intf check failed for chanspec: 0xeb32
Dec  1 19:02:02 acsd: acs_candidate_score_bgnoise(1500): eth6: bgnoise check failed for chanspec: 0xeb32
Dec  1 19:02:02 acsd: acs_candidate_score_txop(1758): eth6: txop check failed for chanspec: 0xeb32

qqw1231, 2022-12-01:
> ok how can you plese write down how I can reproduce it? do you also have logs?

The router cannot be opened in the background due to disconnection. The mobile phone must turn off the wifi and turn it back on, so that the network can be connected again

gnuton, 2022-12-08:
this may happen because you set select auto channel and you have a lot of noise from other 2.5 Ghz network
Do you still get this issue if you select a chanel?

qqw1231, 2022-12-09:
> 

I'm separated from wifi2.4 and wifi5g

> this may happen because you set select auto channel and you have a lot of noise from other 2.5 Ghz network Do you still get this issue if you select a chanel?

I'm separated from wifi2.4 and wifi5g

qqw1231, 2022-12-09:
> this may happen because you set select auto channel and you have a lot of noise from other 2.5 Ghz network Do you still get this issue if you select a chanel?

I'm separated from wifi2.4 and wifi5g

gnuton, 2022-12-09:
[image]
does this problem go away if you select a channel?