gnuton/asuswrt-merlin.ng #298: Killswitch for Wireguard

Nils3211, 2022-12-11:
Can you please add KillSwitch for Wireguard, as you did for OpenVPN.
Im running a TUF-AX5400 with your latest beta, 388.1_0-gnuton0_beta1

Best regards Nils
[image]

gnuton, 2022-12-16:
These features must come from upstream merlin

djt7787, 2022-12-23:
Comment from Merlin on SNB Forums;

WireGuard is not a real client/server setup, it's a peer-to-peer type of setup. WireGuard has no concept of a "disconnected client", the interface will stay up until you manually shut it down, regardless of whether the two are able to communicate or not. If the peers are no longer able to communicate, then the traffic will automatically fail to route without the need of a dedicated killswitch.

https://www.snbforums.com/threads/asuswrt-merlin-388-1-is-now-available-for-all-supported-wifi-6-models.82084/post-805517

Nils3211, 2022-12-25:
Great 👍

closed 2022-12-25 by Nils3211