gnuton/asuswrt-merlin.ng #302: 5Ghz WIFI unstable in 388.1_0-gnuton0_beta2

Rovasteen, 2022-12-22:
TUF-AX5400 flashed to 388.1_0-gnuton0_beta2.  WIFI in the 5Ghz band drops, reconnects and drops again.  Seems cyclic.

Using default professional settings.  Basic settings are:

- 20/40/80/160 MHz Auto
- tick box set for Enable 160 MHz
- Control channel Auto
- tick box set for Auto select channel including DFS channels
- Extension channel Auto
- Authentication Method WPA2/WPA3 Personal
- WPA Encryption AES

After disconnecting over and over again, flashed back to 386.07_2-gnuton1 which is stable.  5Ghz WIFI immediately stable on 386.07_2-gnuton1. 

2.4 Ghz seems stable on both 388.1_0-gnuton0_beta2 and 386.07_2-gnuton1.

The fault reported is unstable WIFI in the 5Ghz band on the 388.1_0-gnuton0_beta2 build.  WIFI in the 5Ghz band drops, reconnects and drops again.

gnuton, 2022-12-22:
do you have any log?

qqw1231, 2022-12-22:
TUF AX 3000 also appeared you this problem, said no one fixed

GaselK, 2022-12-23:
I have zero problems with 5Ghz. (Beta 2). Maybe you should try manual channel setting,

ThrnSC87, 2022-12-23:
No issue here on tuf ax5400 running on a ax client and a few ac devices no issue whatsoever.

Rovasteen, 2022-12-23:
Both 386.07_2-gnuton1 and 388.1_0-gnuton0_beta2 are using Broadcom wireless drivers 17.10 RC157.2809 so it probably isn't the build.  I will play around with this and reopen if I can recreate..

closed 2022-12-23 by Rovasteen

Williams789, 2022-12-25:
This may seem a little different from the above, but in my case, when ISP ip spoofing was enabled, the Wi-Fi device could not be downloaded or slowed down to 800 kbps.(wireless LAN devices have a faster normal range speed)

Both 386.07_2-gnuton1 and 388.1_0-gnuton0_beta2 are the same in my tuf-ax5400.