gnuton/asuswrt-merlin.ng #303: 🌟 Support for ZenWiFi ET8 - RT-AXE95Q 🌟

gnuton, 2022-12-23:
This thread is about supporting RT-AX92U.
If you see this issue here is because I have decided to support this model.

So, feel free to ask questions here about this topic.
I will keep the status updated.

Status:
- [x] received GPL
- [x] received development unit
- [x] Merged GPL to feature-branch
- [x] Initial testing & Bugfixing phase
- [x] Public pre-alphas
- [x] Alpha released
- [x] collecting testing reports  & Bugfixing phase
- [x] Beta released
- [] collecting testing reports  & Bugfixing phase
- [] Stable released

Download firmware: 
PLEASE NOTE: version number stays 386.8 for all these testing images. 
- [Beta 1](https://output.circle-artifacts.com/output/job/0e5e35f5-a26a-4441-88dc-f97dd7eb8dce/artifacts/0/workspace/release/RT-AXE95Q_388.1_0_puresqubi.w.md5)

 
Bugs:
- [x] Wrong image in http://router.asus.com/Advanced_OperationMode_Content.asp

ryanherman, 2022-12-25:
Working great!  Thank you.  I had to modify the download firmware link  and remove the .md5 from the end to download it btw.

gnuton, 2022-12-25:
Beta 2 released 
Fix:
- icons in wifi radar

eric-pierce, 2022-12-27:
Amazing to see another Zenwifi model supported! Selfishly hoping that at the ET12 is similar to the ET8 you may be interested in supporting it as well. Thanks for all your work!

quangkieu, 2022-12-27:
Hey I using ET8 with the XT8. But I got problem setting up the wifi 5GHz. I try using one channel for both systems.
I know the XT8 have 2 antennas but they only support 2 group of channels, 36-64 on 1st and 149-177 on 2nd. So I am trying to use 36+80 band so I only using the 1st antena as the 2nd antena would mixup setting with the ET8 6GHz setting
So I wonder if you could enable using 36+80 band on the ET8 5GHz

nitisha2017, 2022-12-28:
So this beta 2 supports et8? I'll try and let you know how it is on my 4 et8s!

gnuton, 2022-12-28:
@quangkieu AFAIK I cannot do that since all wireless drivers are closed source.

gnuton, 2022-12-28:
@eric-pierce I do not have  any plan for X/ET12 routers

quangkieu, 2022-12-28:
> @quangkieu AFAIK I cannot do that since all wireless drivers are closed source.

Oh, i mean the drop-down list in selection for 5GHz with 80Hz band. So when I set Auto, i know it jump around, sometimes to 36, sometimes to 149, but I could not set fixed to channel 36
[image]

CHF-UK, 2022-12-30:
This is absolutely fantastic.   

I have 4 ET8's and have struggled with issues for the last 18 months.  Previously lots of problems with mesh link stability...6ghz>5.8ghz>6ghz>2.4ghz and general client issues.   It's only been 24 hours with this firmware but even in beta i'm seeing massive improvements compared to ASUS stock firmware.   All uplinks are nailed to 6ghz band and i'm seeing more concurrently connected clients than i've ever seen. The lack of wifi rage from the kids rooms suggests performance for them is good too.

Just a couple of minor issues so far...  Adaptive QoS / Web History seems to crash the GUI which recovered maybe 10 minutes later with similar event in the logs :
Dec 30 11:49:25 kernel: htb: htb qdisc 10: is non-work-conserving?
Dec 30 11:50:26 HTTPD: waitting 10 minitues and restart
Dec 30 11:50:26 rc_service: httpd 18320:notify_rc restart_httpd
Dec 30 11:50:26 ZenWiFi_ET8: start https:8443
Dec 30 11:50:26 ZenWiFi_ET8: start httpd:80
Dec 30 11:50:26 httpd: Save SSL certificate...8443
Dec 30 11:50:26 httpd: mssl_cert_key_match : PASS
Dec 30 11:50:26 httpd: Succeed to init SSL certificate...8443
Dec 30 11:50:26 httpd: Succeed to init SSL certificate...80

Also seem to be struggling with Wireless / WiFi Radar with no output on any of the pages / scans but still have more testing to do there. 

Great work though @gnuton, really appreciate you supporting this model.

gnuton, 2023-01-01:
@CHF-UK 
This is not a httpd crash. It's a planned restart for some reason and that happened becose something has called restart_httpd
from the log it looks like the restart was triggered by SSL certs renewal
Dec 30 11:50:26 HTTPD: waitting 10 minitues and restart 

As for Wifi Radar, that works, but you have to know how it works -_-.. which is not intutuitive at all.
WiFi Radar shows data if in the config enable the collection of the data

gnuton, 2023-01-05:
I merged this code in the master branch!
So I will close this issue. if there are new issues with this model please open a new ticket.

commit f755291b (2019-08-12): nano: updated to 4.2 (closes #303)
closed 2023-01-05 by gnuton