gnuton/asuswrt-merlin.ng #306: LED "Off" Setting ignored in Mesh nodes after mesh system reboot.

evan-d-s, 2022-12-27:
**Router Model Affected**
Models: RT-AX92U, TUF-AX3000

**Firmware Version Affected**
386.08-gnuton1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
When the Router (central node) is scheduled to reboot and sends a restart to mesh nodes the LED settings are ignored, so when the mesh nodes restart the LEDs are always On even if the original setting is "Off". The web GUI and the app report the setting as being "Off", yet the LEDs are on. The setting is ignored even after reconfiguring for LEDs to "Off" (you need to turn them on and back off from the app/GUI) in the next reboot.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure Mesh Node(s) with LED "Off".
2. (Schedule) restart the Router (central node).

**Expected behavior**
The LED "Off" setting should be adhered to even after mesh node reboots.

argh0815, 2022-12-28:
I used to have the same issue on a TUF-AX5400 with 386.07_2-gnuton1. All my RT-AC68U nodes were running on Asuswrt Merlin. After updating the nodes to the latest Asuswrt stock firmware, the LED setting was adhered to.

evan-d-s, 2023-01-02:
So the solution is to revert back to stock? Uninspiring to say the least.

argh0815, 2023-01-02:
The solution that worked for me is as written: leave the router on Gnuton, upgrade the nodes to stock. You basically don't lose any functionality, except if you wanted to run scripts on your nodes. It also has been recommended multiple times over at snbforums to run router on Merlin, nodes on stock.

argh0815, 2023-01-02:
The solution that worked for me is as written: leave the router on Gnuton, upgrade the nodes to stock. You basically don't lose any functionality, except if you wanted to run scripts on your nodes. It also has been recommended multiple times over at snbforums to run router on Merlin, nodes on stock.

If for whatever reason you want to keep the nodes on Gnuton/Merlin, you can try the approach with custom scripts on the nodes as described here: https://www.snbforums.com/threads/aimesh-386-1_2-issue-with-led-lights-on-routers.70823/#post-669323