gnuton/asuswrt-merlin.ng #307: RT-AX92U temperature graph - incorrect line and missing 5-2 radio temperature

mmurza, 2022-12-28:
**Router Model Affected**
Models: RT-AX92U

**Firmware Version Affected**
386.08_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
?

**Describe the bug**
Temperature graph shows 3 temperatures and graphs (at [Administration > Temperatures](http://router.asus.com/Advanced_PerformanceTuning_Content.asp)):
2.4Ghz, 5GHz and CPU
Issue 1: RT-AX92U has 3 WiFi radios, and all 3 temperatures (+CPU) could be displayed, as per [http://router.asus.com/ajax_coretmp.asp](http://router.asus.com/ajax_coretmp.asp)
Issue 2: Current 5Ghz temperature graph is drawn incorrectly: 39C 5Ghz blue line should be below 47C CPU green line. Instead, it's drawn above. And line seems to be drawn constant and smooth (vs slight normal fluctuation of other temperature lines)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to router web interface http://router.asus.com
2. Click on 'Administration > Temperature' (http://router.asus.com/Advanced_PerformanceTuning_Content.asp)
3. See graph
4. In addition, see http://router.asus.com/ajax_coretmp.asp for available temps (3 radios + CPU)

**Expected behavior**
3x Wifi temperatures (2.4, 5-1, 5-2) and CPU temperature should be displayed (and graphs)
Graphs should correspond to data

**Screenshots**
Graph (note blue constant line, above green one):
[image]
Available temperatures (curr_coreTmp_2_raw = "36°C", which is not displayed in the GUI):
[image]
WiFfi temperatures also available from ssh:
wl -i eth5 phy_tempsense | awk '{print $1 / 2 + 20}'
wl -i eth6 phy_tempsense | awk '{print $1 / 2 + 20}'
wl -i eth7 phy_tempsense | awk '{print $1 / 2 + 20}'

gnuton, 2022-12-29:
is this bug still present in the latest 388 beta?

mmurza, 2022-12-30:
Thanks for the note. Issue 2 seems to be fixed in 388 beta.
Graph lines now are in correct order, according the values:
[image]

mmurza, 2022-12-30:
Oh, by the way - found that [Tools > Sysinfo (Temperatures)](http://router.asus.com/Tools_Sysinfo.asp)
Indeed, displays all 4 temperatures, as expected (on 388 beta2)
[image]

Snid0, 2023-01-20:
This issue still exists in the latest stable build 388.1_0-gnuton1

[image]

I put a fan in front of the router the CPU temperature started to drop by the wireless ones stays as they were.

Its worth pointing out the numerical temps under the graph are correct its just the graph which isnt updating.

On a side point is there anyway to get a graph of the temperatures from aimesh nodes?