gnuton/asuswrt-merlin.ng #309: ZenWiFi AX (X8) / RT-AX95Q WiFi Backhall doesn't connect via 5ghz on when the node is updated to 388.1-gnuton0_beta2

shiang-luong, 2022-12-30:
**Router Model Affected**
Models: ZenWiFi AX (X8) / RT-AX95Q

**Firmware Version Affected**
388.1-gnuton0_beta2 

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
Ref post here:
https://www.snbforums.com/threads/fork-dsl-testers-gnutons-alpha-beta-merlin-builds.71150/post-812187

After going from firmware 386.07_2-gnuton1 to 388.1-gnuton0_beta2 on my aimesh node, the wireless back-haul goes from "Great" to "Weak".  The main node is on (388.1-gnuton0_beta2).  Looking at the wireless logs, the biggest difference between the firmware is that on 388.1-gnuton0_beta2 the node doesn't connect via 5ghz.

WiFi Logs:
Current Version : 386.07_2-gnuton1

> SSID: "C71D3***************************************"
> noise: -85 dBm Channel: 116/160
> BSSID: D4:********** Capability: ESS RRM
> Supported Rates: [ 6(b) 9 12 18 24(b) 36 48 54 ]
> HE Capable:
> Chanspec: 5GHz channel 114 160MHz (0xec72)
> Primary channel: 116
> HT Capabilities: 40Mhz SGI20 SGI40
> Supported HT MCS : 0-31
> Supported VHT MCS:
> NSS1 Tx: 0-11 Rx: 0-11
> NSS2 Tx: 0-11 Rx: 0-11
> NSS3 Tx: 0-11 Rx: 0-11
> NSS4 Tx: 0-11 Rx: 0-11
> Supported HE MCS:
> 80 Mhz:
> NSS1 Tx: 0-11 Rx: 0-11
> NSS2 Tx: 0-11 Rx: 0-11
> NSS3 Tx: 0-11 Rx: 0-11
> NSS4 Tx: 0-11 Rx: 0-11
> 160 Mhz:
> NSS1 Tx: 0-11 Rx: 0-11
> NSS2 Tx: 0-11 Rx: 0-11
> NSS3 Tx: 0-11 Rx: 0-11
> NSS4 Tx: 0-11 Rx: 0-11
> 
> Interference Level: Acceptable
> Mode : AP Only
> 
> DFS status: state In-Service Monitoring(ISM) time elapsed 600900ms radar channel cleared by DFS channel 116/160 (0xEC72)
> 
> Channel Information
> ----------------------------------------
> Channel 100 A Band, RADAR Sensitive
> Channel 104 A Band, RADAR Sensitive
> Channel 108 A Band, RADAR Sensitive
> Channel 112 A Band, RADAR Sensitive
> Channel 116 A Band, RADAR Sensitive
> Channel 120 A Band, RADAR Sensitive
> Channel 124 A Band, RADAR Sensitive
> Channel 128 A Band, RADAR Sensitive
> Channel 132 A Band, RADAR Sensitive, Passive
> Channel 136 A Band, RADAR Sensitive, Passive
> Channel 140 A Band, RADAR Sensitive, Passive
> Channel 144 A Band, RADAR Sensitive, Passive
> Channel 149 A Band
> Channel 153 A Band
> Channel 157 A Band
> Channel 161 A Band
> Channel 165 A Band
> Channel 169 A Band
> Channel 173 A Band
> Channel 177 A Band
> 
> Stations List
> ----------------------------------------
> idx MAC Associated Authorized RSSI PHY PSM SGI STBC MUBF NSS BW Tx rate Rx rate Connect Time
> D4:5D:**:**:**:** Yes Yes -30dBm ax No Yes Yes Yes 4 160M 3242.6M 2722.2M 00:07:09

388.1-gnuton0_beta2

> SSID: "C71D3***************************************"
> noise: -85 dBm Channel: 116/160
> BSSID: D4:******* Capability: ESS RRM
> Supported Rates: [ 6(b) 9 12 18 24(b) 36 48 54 ]
> HE Capable:
> Chanspec: 5GHz channel 114 160MHz (0xec72)
> Primary channel: 116
> HT Capabilities: 40Mhz SGI20 SGI40
> Supported HT MCS : 0-31
> Supported VHT MCS:
> NSS1 Tx: 0-11 Rx: 0-11
> NSS2 Tx: 0-11 Rx: 0-11
> NSS3 Tx: 0-11 Rx: 0-11
> NSS4 Tx: 0-11 Rx: 0-11
> Supported HE MCS:
> 80 Mhz:
> NSS1 Tx: 0-11 Rx: 0-11
> NSS2 Tx: 0-11 Rx: 0-11
> NSS3 Tx: 0-11 Rx: 0-11
> NSS4 Tx: 0-11 Rx: 0-11
> 160 Mhz:
> NSS1 Tx: 0-11 Rx: 0-11
> NSS2 Tx: 0-11 Rx: 0-11
> NSS3 Tx: 0-11 Rx: 0-11
> NSS4 Tx: 0-11 Rx: 0-11
> 
> Interference Level: Acceptable
> Mode : AP Only
> 
> DFS status: state In-Service Monitoring(ISM) time elapsed 216600ms radar channel cleared by DFS channel 116/160 (0xEC72)
> 
> Channel Information
> ----------------------------------------
> Channel 100 A Band, RADAR Sensitive
> Channel 104 A Band, RADAR Sensitive
> Channel 108 A Band, RADAR Sensitive
> Channel 112 A Band, RADAR Sensitive
> Channel 116 A Band, RADAR Sensitive
> Channel 120 A Band, RADAR Sensitive
> Channel 124 A Band, RADAR Sensitive
> Channel 128 A Band, RADAR Sensitive
> Channel 132 A Band, RADAR Sensitive, Passive
> Channel 136 A Band, RADAR Sensitive, Passive
> Channel 140 A Band, RADAR Sensitive, Passive
> Channel 144 A Band, RADAR Sensitive, Passive
> Channel 149 A Band
> Channel 153 A Band
> Channel 157 A Band
> Channel 161 A Band
> Channel 165 A Band
> Channel 169 A Band
> Channel 173 A Band
> Channel 177 A Band
> 
> Stations List
> ----------------------------------------
> idx MAC Associated Authorized RSSI PHY PSM SGI STBC MUBF NSS BW Tx rate Rx rate Connect Time

**To Reproduce**
Steps to reproduce the behavior:
1. Update the main node to 388.1-gnuton0_beta2 and node is on 386.07_2-gnuton1.  Assure that the wireless backhaul is still categorized as Great
2. Update the mesh node to 388.1-gnuton0_beta2  (both main and node is now on 388.1-gnuton0_beta2).  
3. Reboot for good measure
4. See that the backhaul now shows "Weak" connect.

**Expected behavior**
The wireless backhaul should remain categorized as "Great" since nothing changed configuration or position wise

**Screenshots**
WiFi Configs:

[image]
[image]

gnuton, 2023-01-06:
can you please reset to factory both and try again?

spysiuk, 2023-01-09:
I had similar issue after going from beta2 to beta3 (RT-AX82U as main router + RT-AX82U and RT-AX92U as AiMesh nodes). I did factory reset after installing beta1 not after beta2 or beta3. Today I did another factory reset on all devices, configured everything from scratch and so far I have great 5G connection.

shiang-luong, 2023-01-15:
@spysiuk and @gnuton thanks!  I will try

closed 2023-01-15 by shiang-luong