gnuton/asuswrt-merlin.ng #310: ZenWiFi AX (X8) / RT-AX95Q seems that only 3 CPU cores show usage, 4th core is always 0%

felalex, 2023-01-01:
**Router Model Affected**
Models: RT-AX95Q

**Firmware Version Affected**
388.1-gnuton0_beta2

**Is this bug present in upstream Merlin releases too?**
Don't know

**Describe the bug**
I've upgraded from the previous 386 release from this repo.
Upgrade was fast and swift (no factory reset).
The CPU graph in the web ui, shown only 3 cores being used, the 4th core is always 0%.

[image]

running stop via ssh shows that the last CPU is being used less than others, but I do see  0% to ~1.5% usage.
Web UI still shows 0%.

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to we UI, see CPU usage.
2. Compare to top from ssh.

**Expected behavior**
Web UI shows the real CPU usage per core.

**Screenshots**
Screenshots are attached.

dacootmeister, 2023-01-07:
Indeed, I have determined the same behaviour with 388.1-gnuton0_beta2
Going to flash 388.1-gnuton0_beta3 in a coupe of hrs

felalex, 2023-01-08:
> Indeed, I have determined the same behaviour with 388.1-gnuton0_beta2 Going to flash 388.1-gnuton0_beta3 in a coupe of hrs
> 
> Update: same issue still exists in 388.1-gnuton0_beta3, although I don't think it's a GUI issue but more fundamental as Core4 is almost not / never utilized at all.
> 
> [image]

Yes, it's used far less, which may not be an issue.
I don't know what may be the implications of utilizing the 4th core in a more balanced way, I guess this is up to ASUS devs to decide which CPU scheduler to use...

Anyways, upgrading to beta 3 didn't help, the behavior is the same.

LarryRosen108, 2023-01-08:
I also don't think it's gnuton related. I remember cpu4 barely being used in official firmware. And I do see it flip to 1% occasionally.
[image]

rmariboe, 2023-01-11:
I'm seeing sporadic core 4 usage in GUI so definitely no UI bug :)

felalex, 2023-01-11:
> I'm seeing sporadic core 4 usage in GUI so definitely no UI bug :)

If you also have XT8, then it means that the behavior is not consistent across devices. The fact that while a cpu is being used, though in smaller percentage, the UI is showing a consistent 0% is indeed a UI bug… maybe the ui update happened when the cpu was indeed 0%, so it’s just that.

rmariboe, 2023-01-12:
> If you also have XT8, then it means that the behavior is not consistent across devices. The fact that while a cpu is being used, though in smaller percentage, the UI is showing a consistent 0% is [...]

Perhaps 🤔 UI on mine appears to correlate with what top is showing, though it's hard to tell exactly due to the slow refresh rates of both... Also XT8.
If you get a consistent 0% in UI while top is showing sustained >0%, that is of course incorrect. If top is only sporadically showing  ~1%, however, I suspect the UI could be averaging that out to 0% rounded down..?

steven0161, 2023-01-24:
I have a XT8 and have installed the stable 388.1_0-gnuton1 and performed a hard factory reset. I have also noticed the core 4 is at zero. The only way I have got it to move slightly is to move a file onto a usb pen drive attached the the XT8.
[image]

rlfrank165, 2023-01-24:
I have also noticed the same thing, even when during router initialization
when the other three cores are all pegged close to 100%.

While certainly not critical does seem a little strange that the CPU
scheduler isn't taking availability of the 4th core to any great degree. Is
that core in any way different from the others in terms of its
functionality or capabilities?  Would be a little strange if it was.

Randy

rlfrank165, 2023-01-24:
only other weird thing is that this function (process scheduling) is so
central to the basic Linux kernel operation (unless the kernel has a
modified scheduler that for some reason is reserving a core for certain
specialized (realtime?) uses, that it's almost hard to see how anything
even remotely dealing with its use as a router could have anything to do
with this, unless this is simply misreporting of what's really going on by
the ASUS UI code.

But this has nothing to do with the Merlin or Gnuton versions.  For a while
I'm running the latest standard ASUS release (388.22068) which is based on
ASUS code more recent than my understanding is the GPL version on which the
current Merlin and Gnuton releases are based.  I'm really trying to see if
these instabilities some of us are seeing are still present in the latest
official ASUS release.  And in the latest ASUS release core 4 is likewise
almost never used.  Will report on my experiences with the ASUS release
when I have some data.

Several months ago when the first 388 ASUS release came out I had upgraded
to it, and it was incredibly unreliable and crashed all of the time, which
is why I switched to Merlin.  I simply want to get some experience with the
latest 388 release from ASUS.

Randy

Randy

Randy

felalex, 2023-01-25:
@gnuton I think that you can remove the beta label if it makes any difference… I also see this in the stable version.

gnuton, 2023-01-25:
I will focus on the beta issue. So i will keep the label. This was a minor one so it just got lower priority.

d3v3l15h, 2023-01-28:
Same problem here.

In fact, 388.1_0-gnuton1 is completely f... up on XT8, same as latest Asus firmwares. that's even why I installed 386 gnuton firmwares in the first place few months ago. I just tried 388.1_0-gnuton1 today, just in case, but reverted back immediately to 386. I have the exact same symptom than with Asus firmware : Aimesh node has very weak 2.4 ghz wireless connection (almost always disconnected in fact) while on 386 it's steadily connected on 5Ghz 2. As soon as I revert back both nodes on 386, everything is perfect again.

gnuton, 2023-01-30:
@d3v3l15h These usually are the kind of issues you get when you dirty flash the router

d3v3l15h, 2023-01-30:
Mmm... Let say the number of times I swapped firmwares, reseted everything and so on, it should have been working at least one time, no ?

ASUS ZenWiFi AX (XT8) Firmware version 9.0.0.4.388.21101
Improved 5GHz backhaul stability.