gnuton/asuswrt-merlin.ng #311: Slow GUI

micron888, 2023-01-02:
TUF-AX5400. anyone having issues with beta2?

[388.1_0-gnuton0_beta2: 
slow gui. cfg_server high memory load. 
factory reset, clear nvram. same slow speed

[388.1_0-gnuton0_beta1: ok, no issue\

swiffer, 2023-01-02:
also running TUF-AX5400 on beta2. Today I've had a lot these log entries. Looks good again after reboot:

dnsmasq-dhcp[1698]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device

gnuton, 2023-01-05:
@micron888 if you check with top or htop the resources can you see any anomaly?

micron888, 2023-01-05:
I have gone back to 386, but based on earlier investigation. It seem to be Cfg_server with high memory load.

gnuton, 2023-01-05:
@micron888 did you have any special setting or anything which can help me to reproduce the issue?

gnuton, 2023-01-05:
@swiffer try to run tree to check what has used so much space on /var.. logs?

micron888, 2023-01-09:
No special settings
Tuf 5400 with 1 aimesh node, xd4
Around 30devices. 
Addon adguard only

gnuton, 2023-01-17:
@micron888 have you restored the setting from a backup after cleaning?
do you get high mem usage when not using scripts?
can you show me the mem usage by app?